open62541/open62541 #1964: missing UA_Client_Subscriptions_addMonitoredItem [help wanted] [client]

fabian727, 2018-08-07:
I am missing the function UA_Client_Subscriptions_addMonitoredItem in the active master-branch/ release. It was here included, now it is missing: [https://github.com/open62541/open62541/blob/d5317cb2e590fab56adbbef75fd5a6255a954fde/include/ua_client_highlevel.h#L603](https://github.com/open62541/open62541/blob/d5317cb2e590fab56adbbef75fd5a6255a954fde/include/ua_client_highlevel.h#L603)

Is there a reason or alternative path to add Items to a subscription?

StalderT, 2018-08-07:
You can check the example here : 

https://github.com/open62541/open62541/blob/59a3cfb8567abefe3f8f2f2f8bddfcf45f32e391/examples/client_subscription_loop.c#L74-L83

fabian727, 2018-08-07:
but there is for each monitored node a single subscription
[https://github.com/open62541/open62541/blob/59a3cfb8567abefe3f8f2f2f8bddfcf45f32e391/examples/client_subscription_loop.c#L64-L83](https://github.com/open62541/open62541/blob/59a3cfb8567abefe3f8f2f2f8bddfcf45f32e391/examples/client_subscription_loop.c#L64-L83)

StalderT, 2018-08-07:
You can call ```UA_Client_MonitoredItems_createDataChange``` multiple times for the same subscription. You can store ```response.subscriptionId``` value (while the subscription is valid).

jpfr, 2018-08-08:
The old API was deprecated for 0.3 and removed for 0.4.

You can use UA_Client_MonitoredItems_createDataChanges to add many monitored items at once.
The relevant header file is comparably short and documented...

closed 2018-08-08 by jpfr