open62541/open62541 #1966: Pointer error when implementing certificate authentication for client

jhj0955, 2018-08-08:
hi

I modified ua_client_connect.c example source for user authentication.
I Added userTokenSignature algorithm and signature to request.
After the request was completed, it was received by the server.
An error occurred while running // UA_ActivateSessionRequest_deleteMembers from the client.
Error - double free or corruption (! prev) : 0x00035fd0.

<pre>
UA_StatusCode
signActivateSessionRequest(UA_SecureChannel *channel,
                           UA_ActivateSessionRequest *request) {

     if(channel == NULL || request == NULL)
         return UA_STATUSCODE_BADINTERNALERROR;

     if(channel->securityMode != UA_MESSAGESECURITYMODE_SIGN &&
        channel->securityMode != UA_MESSAGESECURITYMODE_SIGNANDENCRYPT)
         return UA_STATUSCODE_GOOD;

    const UA_SecurityPolicy *const securityPolicy = channel->securityPolicy;
    UA_SignatureData *signatureData = &request->clientSignature;

   
    /* Prepare the signature */
    size_t signatureSize = securityPolicy->certificateSigningAlgorithm.
                           getLocalSignatureSize(securityPolicy, channel->channelContext);
    UA_StatusCode retval = UA_String_copy(&securityPolicy->certificateSigningAlgorithm.uri,
                                          &signatureData->algorithm);

    retval = UA_ByteString_allocBuffer(&signatureData->signature, signatureSize);
  

    /* Allocate a temporary buffer */
    size_t dataToSignSize = channel->remoteCertificate.length + channel->remoteNonce.length;

    UA_ByteString dataToSign;
    retval = UA_ByteString_allocBuffer(&dataToSign, dataToSignSize);

    /* Sign the signature */
    memcpy(dataToSign.data, channel->remoteCertificate.data, channel->remoteCertificate.length);
    memcpy(dataToSign.data + channel->remoteCertificate.length,
           channel->remoteNonce.data, channel->remoteNonce.length);
    retval = securityPolicy->certificateSigningAlgorithm.
             sign(securityPolicy, channel->channelContext, &dataToSign,
                  &signatureData->signature);

    request->userTokenSignature.signature = signatureData->signature;

    /* Clean up */
    UA_ByteString_deleteMembers(&dataToSign);
    
    return retval;
}
</pre>

<pre>
static UA_StatusCode
activateSession(UA_Client *client) {
 
    UA_ActivateSessionRequest request;
    UA_ActivateSessionRequest_init(&request);
    request.requestHeader.requestHandle = ++client->requestHandle;
    request.requestHeader.timestamp = UA_DateTime_now();
    request.requestHeader.timeoutHint = 600000;

    //manual ExtensionObject encoding of the identityToken
    if(client->authenticationMethod == UA_CLIENTAUTHENTICATION_NONE) {
        UA_AnonymousIdentityToken* identityToken = UA_AnonymousIdentityToken_new();
        UA_AnonymousIdentityToken_init(identityToken);
        UA_String_copy(&client->token.policyId, &identityToken->policyId);
       
        request.userIdentityToken.encoding = UA_EXTENSIONOBJECT_DECODED;
        request.userIdentityToken.content.decoded.type = &UA_TYPES[UA_TYPES_ANONYMOUSIDENTITYTOKEN];
        request.userIdentityToken.content.decoded.data = identityToken;
     
    } 
    else if (client->authenticationMethod == UA_CLIENTAUTHENTICATION_USERNAME) {
        UA_UserNameIdentityToken* identityToken = UA_UserNameIdentityToken_new();
        UA_UserNameIdentityToken_init(identityToken);
        UA_String_copy(&client->token.policyId, &identityToken->policyId);     
        UA_String_copy(&client->username, &identityToken->userName);
        UA_String_copy(&client->password, &identityToken->password);
        request.userIdentityToken.encoding = UA_EXTENSIONOBJECT_DECODED;
        request.userIdentityToken.content.decoded.type = &UA_TYPES[UA_TYPES_USERNAMEIDENTITYTOKEN];
        request.userIdentityToken.content.decoded.data = identityToken;
    }
    else if (client->authenticationMethod == UA_CLIENTAUTHENTICATION_CERTIFICATE) {
        UA_X509IdentityToken* identityToken = UA_X509IdentityToken_new();
     
        UA_X509IdentityToken_init(identityToken);
        identityToken->policyId = UA_STRING_ALLOC("certificate_basic256");
        identityToken->certificateData = client->usercertification;
        client->token.policyId = UA_STRING_ALLOC("0");
        request.userIdentityToken.encoding = UA_EXTENSIONOBJECT_DECODED;
        request.userIdentityToken.content.decoded.type = &UA_TYPES[UA_TYPES_X509IDENTITYTOKEN];
        request.userIdentityToken.content.decoded.data = identityToken;

        request.userTokenSignature.algorithm = UA_STRING_ALLOC("http://www.w3.org/2001/04/xmldsig-more#rsa-sha256");
        
        request.clientSignature.algorithm = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256");
        request.clientSignature.signature = client->sign_userprivatekey;
    }

    /* This function call is to prepare a client signature */
    if(client->channel.securityMode == UA_MESSAGESECURITYMODE_SIGN ||
       client->channel.securityMode == UA_MESSAGESECURITYMODE_SIGNANDENCRYPT) {
        signActivateSessionRequest(&client->channel, &request);
        
    }

    
    UA_ActivateSessionResponse response;
    __UA_Client_Service(client, &request, &UA_TYPES[UA_TYPES_ACTIVATESESSIONREQUEST],
                        &response, &UA_TYPES[UA_TYPES_ACTIVATESESSIONRESPONSE]);
    
    if(response.responseHeader.serviceResult != UA_STATUSCODE_GOOD) {
        UA_LOG_ERROR(client->config.logger, UA_LOGCATEGORY_CLIENT,
                     "ActivateSession failed with error code %s",
                     UA_StatusCode_name(response.responseHeader.serviceResult));
    }
    
    UA_StatusCode retval = response.responseHeader.serviceResult;
    
    /* error!!  double free or corruption (!prev): 0x00035fd0 *** */
    //UA_ActivateSessionRequest_deleteMembers(&request);
    
    UA_ActivateSessionResponse_deleteMembers(&response);
    
    return retval;
}
</pre>

jpfr, 2018-08-08:
You probably want to make a deep copy of client->usercertification.
In addition, the valgrind tool gives the other code location where the code was freed.

jhj0955, 2018-08-09:
Thank you for your answer.
But I didn't understand your message.
I want to resolve the error when I run the UA_ActivateSessionRequest_deleteMembers(&request).
error - (double free or corruption (! prev) : 0x00035fd0. )

if deleteMembers (request) is not called, the server and client communicate in a user certificate manner.
But what happens if I don't deleteMembers function?

jpfr, 2018-08-09:
Did you run the client with the valgrind tool?

Please do so and post the output.

jhj0955, 2018-08-10:
[image]

execuc, 2020-06-22:
Problem comes from this line `request->userTokenSignature.signature = signatureData->signature;`
At the end, library will want to free the `client signature``and the `token signature` which lead to `Error - double free or corruption `

jpfr, 2023-06-02:
Appears to be fixed.

closed 2023-06-02 by jpfr