open62541/open62541 #1967: How to get a list of all nodes for OPC UA client

dojery, 2018-08-08:
!ATTENTION!
Please read the following page carefully and provide us with all the
information requested:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Use Github Markdown to format your text:
https://help.github.com/articles/basic-writing-and-formatting-syntax/

Fill out the sections and checklist below (add text at the end of each line).

Remove this text between the ATTENTION marks after reading it.
!ATTENTION!
--------------------------------------------------------------------------------

Description
===========

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue
How to get a list of all nodes for OPC UA client?

dojery, 2018-08-08:
How to get a list of all nodes for OPC UA client,when client connected the server?

jpfr, 2018-08-08:
This is not possible.
You can use the browse service to discover the information model.
But this returns only the references to/from a particular node.

closed 2018-08-08 by jpfr

jpfr, 2018-08-09:
Also the .net client can only use the standard service. I.e. browse.

Normally browse is enough.

Of course you can implement a custom tree traversal algorithm for your client.

dojery, 2018-08-09:
how to do a custom tree traversal algorithm for your client？which way can be used？please help me，thanks.

schroeder-, 2018-08-09:
Use UA_Client_Service_browse and go over all Nodes starting UA_NS0ID_OBJECTSFOLDER and go over all Children Folder and Objects.

dojery, 2018-08-09:
can you provide me an example file?I am a new hand,thanks.

schroeder-, 2018-08-09:
See examples/client.c that should you get going.

dojery, 2018-08-09:
This example can only get a first-level name, and the sub-name cannot be obtained.

dojery, 2018-08-09:
QQ：1831974697 ，we can discuss a step further.

dojery, 2018-08-09:
this is a first -level name,but their children can not obtained.