open62541/open62541 #1969: [Windows Build error] CMake build failed

mukunth1987, 2018-08-10:
Description
===========
Cloned the latest code and tried to build as per the instructions.
cmake .. -G "Visual Studio 2015" failed with below error

CMake Error at CMakeLists.txt:36 (string):
  **_string sub-command REPLACE requires at least four arguments._**

-- Could NOT find Sphinx (missing: SPHINX_EXECUTABLE)
-- Could NOT find LATEX (missing: LATEX_COMPILER)
-- Configuring incomplete, errors occurred!

Background Information / Reproduction Steps
===========================================
1) cmake v3.11.4
2) python 3.7
3) Visual Studio 2015

Followed the build instructions. But cmake build failed with :
**_string sub-command REPLACE requires at least four arguments._**

in CMakeLists.txt, error occurs in the below line
string(REPLACE "\n" "" OPEN62541_VER_COMMIT ${GIT_COM_ID} )

but if I add double quotes to the last argument, it works fine.
string(REPLACE "\n" "" OPEN62541_VER_COMMIT "${GIT_COM_ID}" )

Please let me know your opinion on this.

mukunth1987, 2018-08-10:
[UPDATE]
It looks like GIT_COM_ID output variable is empty after 

execute_process(COMMAND ${GIT_EXECUTABLE} describe --always --tags
                    RESULT_VARIABLE res_var OUTPUT_VARIABLE GIT_COM_ID WORKING_DIRECTORY ${CMAKE_SOURCE_DIR})

But if run "git describe --always --tags" in command prompt, it displays the output. But its not stored in GIT_COM_ID.

Any opinion on this ?

Pro, 2018-08-13:
It looks like you are using cmake and git in two different shell environments.
Could it be that cmake is not able to find the git executable in the PATH environment?

We should add an additional check here:
https://github.com/open62541/open62541/blob/808239dc8ac79a7ee9b712a23ddcedd61ee38794/CMakeLists.txt#L34 to handle empty GIT_COM_ID

Pro, 2018-08-13:
Can you try to add the patch from #1975 and see if that solves your issue?
(It will also be merged to master)

mukunth1987, 2018-08-13:
hi @Pro 
I uninstalled git v2.18.0 64-bit installer and installed 32-bit version of the same installer. After that, build worked fine.

commit ac7efc15 (2018-08-23): Add additional check for valid git commit id. Fixes #1969
commit c8cc23e6 (2018-09-05): Add additional check for valid git commit id. Fixes #1969
closed 2018-09-05 by Pro