open62541/open62541 #1971: Subscriptions deleted again immediately?

Flohack74, 2018-08-10:
Description
===========
Shortly after enabling the publishing for a subscription it is being deleted. A first set of item changes can be triggered by calling UA_Client_runAsync immediately after setting the publish, but this just delays the deletion.

Background Information / Reproduction Steps
===========================================
See below the logfile snippet.

Checklist
=========
Please provide the following information:

 - [x] open62541 Version 03.rc2
 - [x] Other OPC UA SDKs used (client or server): None
 - [x] Operating system: Windows 7
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

I got the following log from our application:
```
2018-08-10 14:38:08,424 INFO  Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  OpcUa.Wrapper: Connection to opc.tcp://172.24.252.176:48010 => success
2018-08-10 14:38:08,430 INFO  Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  OpcUa.Wrapper: Namespace lookup for urn:UnifiedAutomation:CppDemoServer:BuildingAutomation => success
2018-08-10 14:38:08,432 DEBUG Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  Server status == Running ('opc.tcp://172.24.252.176:48010').
2018-08-10 14:38:08,441 INFO  Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  OpcUa.Wrapper: Subscription created  => success
2018-08-10 14:38:08,444 DEBUG Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  Added a refresh group 'ModelRead' with name 'ModelRead' ('opc.tcp://172.24.252.176:48010').
2018-08-10 14:38:08,447 INFO  Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  OpcUa.Wrapper: Subscription created  => success
2018-08-10 14:38:08,448 DEBUG Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  Added a refresh group 'ModelWrite' with name 'ModelWrite' ('opc.tcp://172.24.252.176:48010').
2018-08-10 14:38:08,454 INFO  Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  OpcUa.Wrapper: Monitored item created Furnace_1.GasFlow => success
2018-08-10 14:38:08,456 DEBUG Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  Added item [Name='Furnace1GasFlow',OpcName='Furnace_1.GasFlow'] to refresh group 'ModelRead' ('opc.tcp://172.24.252.176:48010').
2018-08-10 14:38:08,458 INFO  Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  OpcUa.Wrapper: Monitored item created Furnace_1.PowerConsumption => success
2018-08-10 14:38:08,460 DEBUG Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  Added item [Name='Furnace1PowerCons',OpcName='Furnace_1.PowerConsumption'] to refresh group 'ModelRead' ('opc.tcp://172.24.252.176:48010').
2018-08-10 14:38:08,463 INFO  Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  OpcUa.Wrapper: Monitored item created Furnace_1.State => success
2018-08-10 14:38:08,464 DEBUG Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  Added item [Name='Furnace1State',OpcName='Furnace_1.State'] to refresh group 'ModelRead' ('opc.tcp://172.24.252.176:48010').
2018-08-10 14:38:08,467 INFO  Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  OpcUa.Wrapper: Monitored item created Furnace_1.Temperature => success
2018-08-10 14:38:08,468 DEBUG Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  Added item [Name='Furnace1TempPV',OpcName='Furnace_1.Temperature'] to refresh group 'ModelRead' ('opc.tcp://172.24.252.176:48010').
2018-08-10 14:38:08,473 INFO  Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  OpcUa.Wrapper: Monitored item created Furnace_1.TemperatureSetPoint => success
2018-08-10 14:38:08,475 DEBUG Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  Added item [Name='Furnace1TempSP',OpcName='Furnace_1.TemperatureSetPoint'] to refresh group 'ModelWrite' ('opc.tcp://172.24.252.176:48010').
2018-08-10 14:38:08,477 INFO  Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  Create items took 26.1412 ms.
2018-08-10 14:38:08,479 DEBUG Com.Andritz.MERe.Common.Servers.L1Server.Opc.OpcModule :  OpcClient 'opc.tcp://172.24.252.176:48010' connected.
2018-08-10 14:38:08,484 INFO  Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  OpcUa.Wrapper: Set publishing mode 2106930690 => success
2018-08-10 14:38:08,486 DEBUG Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  Group 'ModelRead' activated ('opc.tcp://172.24.252.176:48010').
2018-08-10 14:38:08,489 INFO  Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  OpcUa.Wrapper: Deleted subscription 2106930691
2018-08-10 14:38:08,492 INFO  Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  OpcUa.Wrapper: Deleted subscription 2106930690
2018-08-10 14:38:08,494 WARN  Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  OpcUa.Wrapper: Set publishing mode 2106930691 => error 80060000
2018-08-10 14:38:08,497 DEBUG Com.Andritz.MERe.Common.Servers.L1Server.Opc.Ua.OpcUaClient :  Group 'ModelWrite' activated ('opc.tcp://172.24.252.176:48010').
```
* First the connection to the server is established => OK
* 2 Subscriptions are created, one with 500ms, the other with 0ms with initially publish set to false => OK
* 5 Monitored items are added to the subscriptions => OK
* Then the subscription publish modes are set to true => OK

Now here starts the problem, immediately after setting this for the first subscription there is a callback informing about the deletion of both subscriptions.

I miss a bit the documentation for when to use UA_Client_runAsync() and `UA_Client_Subscriptions_manuallySendPublishRequest() `

StalderT, 2018-08-10:
Can you provide a sample client code?

Flohack74, 2018-08-10:
Yes, after the weekend. Basically I am trying to extend the wrapper from here: https://github.com/open62541/open62541/wiki/Using-open62541-from-C%23 for having subscriptions and events accessible from managed code. Or if someone else has already worked on that, I am happy to share/learn ;)

Pro, 2018-08-13:
@Flohack74 note that @bazza2jw already wrote a C++ wrapper for open62541:
https://github.com/bazza2jw/open62541Cpp

Maybe you can support him or have a look at his code to learn more.

Anyways a sample code would be nice to reproduce the problem.

Flohack74, 2019-10-09:
Ok so I am no longer working on this project, therefore have no resources anymore to track or contribute. Sorry for that!

closed 2019-10-09 by Flohack74