open62541/open62541 #1977: Default NodeSet xml file lacks ModellingRules nodes

jonasgreen88, 2018-08-13:
Description
===========
The minimal NodeSet xml file used as default NodeSet by cmake lacks the ModellingRules nodes "Mandatory" and "Optional". According to the OPC specification Part 5, chapter 6.3.2 ServerCapabilitiesType: "_ModellingRules is an entry point to browse to all ModellingRules supported by the Server. All ModellingRules supported by the Server should be able to be browsed starting from this Object._"

Do you guys interpret the specification differently or have the nodes just been forgotten to be added?

https://github.com/open62541/open62541/blob/02734c310dc4f29a45879eeae1e1b7f63d9c9c12/tools/schema/Opc.Ua.NodeSet2.Minimal.xml#L1346

I can also see that the same xml file on 0.3 branch lacks some more mandatory nodes of the ServerObject (AggregateFunctions, ModellingRules, VendorServerInfo). Will the ServerObject nodes added to the NodeSet on the master branch be merged to 0.3 before release?
https://github.com/open62541/open62541/blob/215651ab8db94e5eacdd10ec26a5a9fb96b9301f/tools/schema/Opc.Ua.NodeSet2.Minimal.xml#L1

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 02734c310dc4f29a45879eeae1e1b7f63d9c9c12
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-08-15:
On master, we support three levels of completeness for namespace zero.
Minimal (no generated code), reduced (default) and full.

Which parts of namespace zero we include where is up for debate.
In the future, we might have the `Opc.Ua.NodeSet2.PartNN.xml`  files individually selectable.
https://opcfoundation.org/UA/schemas/1.04/

Feel free to propose a pull request with specific changes.

Pro, 2018-08-21:
Summary from the TelCo: @jonasgreen88 will have a look at it. This feature is already in master and only needs to be merged into 0.3