open62541/open62541 #1981: Segfault when writing large stringArray to OPC UA server using basic256sha256 sign and encrypt

RantiE, 2018-08-15:
Description
===========
Hi,

I have managed to cause a segfault when writing a large stringArray to a separate UA server. 

Background Information / Reproduction Steps
===========================================
I am using a recently build single file distribution based on the master branch. 
I have been trying to send different data types to the Prosys OPC UA simulation server using a modified version of the basic256sha256 example. The code below is an effort to reproduce the potential bug. As far as I can tell the crash happens when trying to write an array of over 1612 elements of a single string character. The program crashes with `free(): invalid size` or `munmap_chunk(): invalid pointer`

I have managed to write stringArrays of a greater size than above using the none/none connection without any issues.
I have attached a copy of my gdb output for running the program and the code to reproduce the error is also pasted below. Please let me know without hesitation if you require any further information.

Thanks in advance for your assistance in resolving this issue.

Checklist
=========
Please provide the following information:

 - [ x ] open62541 Version (release number or git tag): 0.3-rc2-313-g59a3cfb8
 - [ x ] Other OPC UA SDKs used (client or server): Prosys OPC UA Simulation server
 - [ x ] Operating system: Ubuntu 18.04
 - [ x ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: see gdb.txt
 - [ ] Wireshark network dump attached: I can supply this if it is required.
 - [ x ] Self-contained code example attached
 - [ x ] Critical issue
`/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
* See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

#include "open62541.h"
#include "common.h"
#include <pthread.h>

#define MIN_ARGS           3
#define FAILURE            1
#define CONNECTION_STRING  "opc.tcp://rantie:53530/OPCUA/SimulationServer"

/* cleanupClient deletes the memory allocated for client configuration.
*
* @param  client             client configuration that need to be deleted
* @param  remoteCertificate  server certificate */
static void cleanupClient(UA_Client* client, UA_ByteString* remoteCertificate) {
UA_ByteString_delete(remoteCertificate); /* Dereference the memory */
UA_Client_delete(client); /* Disconnects the client internally */
}

void * connectionMonitor(void *);
bool clientConnect();
pthread_t thread0;
UA_Client* client;
int arraySize = 1500;

/* main function for secure client implementation.
*
* @param  argc               count of command line variable provided
* @param  argv[]             array of strings include certificate, private key,
*                            trust list and revocation list
* @return Return an integer representing success or failure of application */
int main(int argc, char* argv[]) {
   client             = NULL;
   UA_ByteString*          remoteCertificate  = NULL;
   UA_StatusCode           retval             = UA_STATUSCODE_GOOD;
   size_t                  trustListSize      = 0;
   UA_ByteString*          revocationList     = NULL;
   size_t                  revocationListSize = 0;

   /* endpointArray is used to hold the available endpoints in the server
   * endpointArraySize is used to hold the number of endpoints available */
   UA_EndpointDescription* endpointArray      = NULL;
   size_t                  endpointArraySize  = 0;

   if(argc < MIN_ARGS) {
      UA_LOG_FATAL(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
         "The Certificate and key is missing."
         "The required arguments are "
         "<client-certificate.der> <client-private-key.der> "
         "[<trustlist1.crl>, ...]");
      return FAILURE;
   }

   /* Load certificate and private key */
    UA_ByteString certificate = loadFile(argv[1]);
    UA_ByteString privateKey = loadFile(argv[2]);
   //UA_ByteString   revocationList   = loadFile("/home/ranti/git/kepware-writer/metadata/certs/lists/..."); //only loading one list atm - need to extend to read all lists in folder
   /* The Get endpoint (discovery service) is done with
   * security mode as none to see the server's capability
   * and certificate */
   client = UA_Client_new(UA_ClientConfig_default);
   remoteCertificate = UA_ByteString_new();
   retval = UA_Client_getEndpoints(client, CONNECTION_STRING,
      &endpointArraySize, &endpointArray);
   if(retval != UA_STATUSCODE_GOOD) {
      UA_Array_delete(endpointArray, endpointArraySize,
         &UA_TYPES[UA_TYPES_ENDPOINTDESCRIPTION]);
      cleanupClient(client, remoteCertificate);
      return (int)retval;
   }

   UA_String securityPolicyUri = UA_STRING("http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256");
   printf("%i endpoints found\n", (int)endpointArraySize);
   for(size_t endPointCount = 0; endPointCount < endpointArraySize; endPointCount++) {

      if(endpointArray[endPointCount].securityMode != UA_MESSAGESECURITYMODE_SIGNANDENCRYPT)
         continue;

      if(UA_String_equal(&endpointArray[endPointCount].securityPolicyUri, &securityPolicyUri)) {
         UA_ByteString_copy(&endpointArray[endPointCount].serverCertificate, remoteCertificate);
         break;
      }
   }

   if(UA_ByteString_equal(remoteCertificate, &UA_BYTESTRING_NULL)) {
      UA_LOG_FATAL(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
         "Server does not support Security Basic256Sha256 Mode of"
         " UA_MESSAGESECURITYMODE_SIGNANDENCRYPT");
      cleanupClient(client, remoteCertificate);
      return FAILURE;
   }

   UA_Array_delete(endpointArray, endpointArraySize,
      &UA_TYPES[UA_TYPES_ENDPOINTDESCRIPTION]);

   UA_Client_delete(client); /* Disconnects the client internally */

   /* Load the trustList. Load revocationList is not supported now */
   if(argc > MIN_ARGS)
      trustListSize = (size_t)argc-MIN_ARGS;

   UA_STACKARRAY(UA_ByteString, trustList, trustListSize);
   for(size_t trustListCount = 0; trustListCount < trustListSize; trustListCount++) {
      trustList[trustListCount] = loadFile(argv[trustListCount+3]);
   }

   /* Secure client initialization */
   //client = UA_Client_new(UA_ClientConfig_default);
   client = UA_Client_secure_new(UA_ClientConfig_default,
      certificate, privateKey,
      remoteCertificate,
      trustList, trustListSize,
      revocationList, revocationListSize,
      UA_SecurityPolicy_Basic256Sha256);
   if(client == NULL) {
      UA_ByteString_delete(remoteCertificate); /* Dereference the memory */
      return FAILURE;
   }

   UA_ByteString_deleteMembers(&certificate);
   UA_ByteString_deleteMembers(&privateKey);
   for(size_t deleteCount = 0; deleteCount < trustListSize; deleteCount++) {
      UA_ByteString_deleteMembers(&trustList[deleteCount]);
   }

   /* Secure client connect */
   retval = UA_Client_connect(client, CONNECTION_STRING);
   if(retval != UA_STATUSCODE_GOOD) {
      cleanupClient(client, remoteCertificate);
      return (int)retval;
   }
   pthread_create(&thread0, NULL, connectionMonitor, NULL);
   pthread_join(thread0, NULL);
}

void * connectionMonitor(void * param)
{
   usleep(100);
   while (true)
   {
      if (UA_Client_getState(client)==UA_CLIENTSTATE_DISCONNECTED)
      {
         bool connected=false;
         for (int i=0;i<5;i++)
         {
            connected = clientConnect();
            if(connected)
            {
               i=5;
            }
            else
            {
               sleep(5);
            }            
         }
      }
      usleep(100);
      //try to write increasingly large string array here
      UA_Variant * myVariant= UA_Variant_new();
      UA_String arr[arraySize];
      for (int i=0; i<arraySize; i++){
         arr[i] = UA_String_fromChars("a");
      }

      UA_Variant_setArrayCopy(myVariant, arr,arraySize, &UA_TYPES[UA_TYPES_STRING]);
      UA_Client_writeValueAttribute(client, UA_NODEID_STRING(3, "StringArray"), myVariant);
      printf("Writing array of size: %d\n", arraySize);
      arraySize++;
   }
   return NULL;
}

bool clientConnect()
{
   UA_StatusCode status = UA_Client_connect(client, CONNECTION_STRING);
   if (status!=UA_STATUSCODE_GOOD)
   {
      return false;
   }
   return true;
}
[gdb.txt](https://github.com/open62541/open62541/files/2290577/gdb.txt)
`

jpfr, 2018-08-15:
@Infinity95 Would you put this on your stack of work?

RantiE, 2018-08-16:
Thanks for looking into this @Infinity95. Can you kindly confirm where the maximum tcp message size configured? I ask so that I can look at ways to potentially avoid exceeding it.

Infinity95, 2018-08-17:
You would have to check in the prosys server manual or something similar. I'm not familiar with that software. In open62541 it would be the recvBufferSize Variable in the config structure. (On the server side)
This should only be a temporary workaround though. The client is broken and we need to fix it.

jpfr, 2018-08-17:
Clarification: Client and server negotatie the sendBufferSize and the maxMessageSize.
sendBufferSize refers to chunks.
maxMessageSize refers to messages that can be assembled from several chunks.

The default client settings are here: https://github.com/open62541/open62541/blob/master/plugins/ua_config_default.c#L673

teccheck, 2025-04-15:
There's been no further activity and a was clarification provided

closed 2025-04-15 by teccheck