open62541/open62541 #1982: About dynamic memory allocation (malloc()/free())

i-and, 2018-08-15:
Does the library use dynamic allocation and freeing of memory after its initialization and in the process of work? If so, do you plan to implement this critical real-time feature in future versions (for example, work in the pre-allocated buffers)?

jpfr, 2018-08-15:
OPC UA is half database and half network protocol.
The database part is near impossible to implement without malloc.
That said, there are two options for malloc-less operations

1. Replace the UA_malloc and UA_free macros with your custom malloc implementation.
https://github.com/open62541/open62541/wiki/Different-memory-allocator

2. TCP/IP is not realtime anyway. So where this counts is for the new PubSub extension, possibly used for time-sensitive networking. See the following publication for our work on realtime PubSub with open62541: https://www.researchgate.net/publication/326710779_Open_Source_OPC_UA_PubSub_over_TSN_for_Realtime_Industrial_Communication

i-and, 2018-08-20:
Do you have any plans to implement in open62541 the OPC UA TSN client and server in the future? As will be supported by configuring the parameters of the TSN messages?

jpfr, 2018-09-30:
There is no TSN client/server at the moment.

We have support for OPC UA PubSub (which is a different protocol) in this repository.
But you have to jump through some hoops to make it work without malloc.
We have done this internally. But this will take some more time before it is ready for public consumption.

closed 2018-09-30 by jpfr