open62541/open62541 #1988: mDNS setServerOnNetworkCallback for Clients

fabian727, 2018-08-17:
Hi guys,
I try to automate my setup a little more. For this I would like to detect my Servers automatically, which can be achieved through the LDS/GDS Servers with MultiCast enabled, which is working very fine. Nice examples.

Though I have one question (maybe @Pro): how are clients supposed to find their server(s), if they do not have the address of the LDS-Server? The only way I found is to implement a server, who searches for mDNS-servers, like in the server_multicast.c example. Why is there no such behaviour for clients? Detect automatically the LDS server. If the client is trustworthy he has a certificate and can connect to it, else you might have a hacker, who doesn't know pretty much more than before.

Thanks,
Fabian

jpfr, 2018-08-17:
An attacker could of course inject himself in the mDNS discovery process.
The GDS provides concepts for building a chain of trust.
But integration in a new (and hostile) environment is tricky.

@ChHaa Has more to say on that topic.

fabian727, 2018-08-17:
> But integration in a new (and hostile) environment is tricky.
can you explain this? Of which integration are you speaking?

At the moment I find it kind of funny, that OPC UA is for automation, but a client is not capable finding a server, except you clearly direct him to it

jpfr, 2018-08-17:
> But integration in a new (and hostile) environment is tricky.

That should read "secure integration". Somehow you need to trust the certificate of the remote side.
Eg. via some form of public key infrastructure.

> At the moment I find it kind of funny, that OPC UA is for automation, but a client is not capable finding a server, except you clearly direct him to it

With mDNS you can find the server without a known hostname.
Similar to how printers are discovered in a network.
mDNS discovery defines a multicast address that is used to disseminate the information.
And the multicast address is always the same.

Pro, 2018-11-06:
It is possible to add a client API for mDNS using the ua_mdns.c file functionality https://github.com/open62541/open62541/blob/master/src/server/ua_mdns.c

That would need some refactoring on both, the client and server side. We are happy for any PR which implements this feature for the client