open62541/open62541 #1991: Full Namespace0: Stack area around _alloca memory reserved by this function is corrupted

crafti5, 2018-08-18:
Hi all,

I tried to use full namespace 0 and got an error with initializing all nodes:
 "Stack area around _alloca memory reserved by this function is corrupted"

I use VS2008 for compiling and I also changed the Stack with "/STACK:8000000". 

Further notice it happens always at the following code snippet:

```
static UA_StatusCode function_ua_namespace0_117_begin(UA_Server *server, UA_UInt16* ns) {
UA_StatusCode retVal = UA_STATUSCODE_GOOD;
UA_VariableAttributes attr = UA_VariableAttributes_default;
attr.minimumSamplingInterval = 0.000000;
attr.userAccessLevel = 1;
attr.accessLevel = 1;
attr.valueRank = -1;
attr.dataType = UA_NODEID_NUMERIC(ns[0], 294);
UA_STACKARRAY(UA_UtcTime, variablenode_ns_0_i_2258_variant_DataContents, 1);
UA_init(variablenode_ns_0_i_2258_variant_DataContents, &UA_TYPES[UA_TYPES_UTCTIME]);
UA_Variant_setScalar(&attr.value, variablenode_ns_0_i_2258_variant_DataContents, &UA_TYPES[UA_TYPES_UTCTIME]);
attr.displayName = UA_LOCALIZEDTEXT("", "CurrentTime");
#ifdef UA_ENABLE_NODESET_COMPILER_DESCRIPTIONS
attr.description = UA_LOCALIZEDTEXT("", "");
#endif
attr.writeMask = 0;
attr.userWriteMask = 0;
retVal |= UA_Server_addNode_begin(server, UA_NODECLASS_VARIABLE,
UA_NODEID_NUMERIC(ns[0], 2258),
UA_NODEID_NUMERIC(ns[0], 2256),
UA_NODEID_NUMERIC(ns[0], 47),
UA_QUALIFIEDNAME(ns[0], "CurrentTime"),
UA_NODEID_NUMERIC(ns[0], 63),
(const UA_NodeAttributes*)&attr, &UA_TYPES[UA_TYPES_UTCTIME],NULL, NULL);
return retVal;
}
```

If I change the Stack parameter it won't change anything.
Does anybody had the same problem with full namepsace0 or is there any solution to my problem?

best regards

crafti5, 2018-08-20:
@Pro 
sure I can, it was line 33965.

unfortunately it was gone, I have done a recompilation and I checked "UA_FILE_NS0" and Namespace 0 "FULL". After unchecking "UA_FILE_NS0" the problem was gone.  Can anyone explain why this can happen or what was happening?

Pro, 2018-08-20:
Unfortunately the line number does not help much since the code is dynamically generated. It would help to have the code on that line.

What do you mean by unchecking `UA_FILE_NS0`?
That's a string value in CMake, so you can only change the value of it.
It could be that now you are not using the full NS0 file anymore and thus you do not run into that problem (UA_FILE_NS0 defines which Nodeset the stack should use, i.e., minimal or full).

Can you verify that `UA_FILE_NS0` is pointing to `open62541/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml`?

jpfr, 2018-08-20:
UA_STACKARRAY resolves to either alloca or a variable length array (`int myArray[myLen];`).
The latter cannot be checked for a NULL return.
And alloca also never returns NULL. It just crashes the program with a stack overflow.

At most we can wrap UA_STACKARRAY into a general maximum length check for stack allocations.
AFAIK we already have length checks around UA_STACKARRAY everywhere but the generated code...

crafti5, 2018-08-20:
@Pro do you mean the exact point where it happens? I saw this issue after returning of the function. Will this help you?

By checking I mean the checkmark in cmake-gui (when advanced view is activated).

Sorry but I can't verify  last question of you. I forgot to save a copy of this issue :-/.
I can try to repodruce this in a few days but currently I have this time.

teccheck, 2025-04-15:
There's been no activity for quite some time and VS2008 is outdated now, so I'll close this issue

closed 2025-04-15 by teccheck