open62541/open62541 #1995: UAExpert does't show events of the server events tutorial

sqadasi, 2018-08-20:
Hello everyone,

I've tried to get the demo from **examples/tutorial_server_events.c** running, however it was not as trivial as expected...

I tested the demo first with UAExpert, which didn't show the event triggered by the UAMethod "Generate Event" in its event view. Testing with other UA Clients (Softing and OPC Foundation Sample Client) I got the triggert event, but only after a RepublishRequest (in about 18 seconds after the method call). 
What I've noticed by the "working clients", is that when I change the filter settings of the Clause Fields, so I only have the EventFields checked which are getting set by the demo program (SourceNode, ReceiveTime, EventId, EventType) by the function **eventSetStandardFields()** from "ua_subscription_events.c" and (Message, Severity) by the function **setUpEvent** from "tutorial_server_events.c", I get the events directly after calling the triggering method.

Unfortunately setting up the filter in UAExpert is restricted, so I added the missing EventField "Time" (property of the BaseEventType) to the function **setUpEvent** and the triggered event is now appearing in UAExpert!

Wouldn't it be better to add all the mandatory EventFields and not only (Message and Severity) to the demo program, so it can be tested directly without the need to modify the demo program or the Client filters?

That could be the same problem as already mentioned in issue #1839.

Best regards

jpfr, 2018-08-20:
> Wouldn't it be better to add all the mandatory EventFields and not only (Message and Severity) to the demo program, so it can be tested directly without the need to modify the demo program or the Client filters?

Obsolutely! Feel free to post a pull request with the change.
Otherwise you can coordinate with @Pro and @aDogCalledSpot to make the example UAExpert compliant.

Pro, 2018-08-20:
@sqadasi thanks for diggint into it!
We will definitely should add all the mandatory fields to the event. Actually @aDogCalledSpot was already looking into the issue that events do not show up in UaExpert. Looks like you found the solution to that issue.

Can you create a PR? Otherwise @aDogCalledSpot will also add a fix ASAP.

PR #1996 (2018-08-21): add EventFields 'Time' and 'SourceName' in server events tutorial
I extended the example to add the rest mandatory EventFields 'Time' and 'SourceName'.
UaExpert Client needs the EventField 'Time' to show the events on Event View.

That fixes #1995 and maybe relevant to issue #1839.

@Pro and @aDogCalledSpot please check it:)
closed 2018-08-21 by Pro