open62541/open62541 #1997: How to use method return values?

NIPE-SYSTEMS, 2018-08-21:
I'm trying to get a method callback work so that it returns values. My callback looks like this:

```cpp
UA_StatusCode my_callback(
    UA_Server *server,
    const UA_NodeId *sessionId,
    void *sessionContext,
    const UA_NodeId *methodId,
    void *methodContext,
    const UA_NodeId *objectId,
    void *objectContext,
    size_t inputSize,
    const UA_Variant *input,
    size_t outputSize,
    UA_Variant *output
) {
    assert(outputSize == 2);
    bool some_bool = true;
    std::string some_string("foo");
    UA_Variant_setScalarCopy(&output[0], &some_bool, &UA_TYPES[UA_TYPES_BOOLEAN]);
    UA_String string = UA_STRING_CPP(some_string);
    UA_Variant_setScalarCopy(&output[1], &string, &UA_TYPES[UA_TYPES_STRING]);
    return UA_STATUSCODE_GOOD;
}
```

Assuming the method types are correct. When the method gets called, a SEGFAULT happens. Am I doing something wrong? How is the intended use of method return values? Are more informations required?

Thanks in advance!

P.S.: I'm at commit hash f6eae36ebe61c666d2b573ce88a50e6f581aa62a.

fabian727, 2018-08-22:
as far as I know you have to allocate the memory for two elements before you can copy with 
UA_Variant_setArray()

after that you can copy different variables and types into it

I made it this way:
```c
output     = new UA_Variant[arrayLength];
UA_Variant_setScalarCopy(&inputArray[x],&some_bool,&UA_TYPES[UA_TYPES_BOOLEAN]);
```

For more info look [here](https://groups.google.com/forum/#!topic/open62541/JOedsy3W5NY
which directs to [here](https://github.com/open62541/open62541/blob/master/include/ua_client_highlevel.h#L396)

NIPE-SYSTEMS, 2018-08-22:
Thanks @fabian727 for your answer!

I think your hint does not work for me since the function prototypes are different:

```cpp
typedef UA_StatusCode (*UA_MethodCallback)(UA_Server *server, const UA_NodeId *sessionId, void *sessionContext, const UA_NodeId *methodId, void *methodContext, const UA_NodeId *objectId, void *objectContext, size_t inputSize, const UA_Variant *input, size_t outputSize, UA_Variant *output);
UA_StatusCode UA_EXPORT UA_Client_call    (UA_Client *client, const UA_NodeId  objectId,                        const UA_NodeId  methodId,                                                                      size_t inputSize, const UA_Variant *input, size_t *outputSize, UA_Variant **output);
```

Notice the last arguments: My callback gives me `size_t outputSize, UA_Variant *output` and the client invocation-function receives `size_t *outputSize, UA_Variant **output`.

That means I need to call

```cpp
*output = new UA_Variant[arrayLength];
```

but this requires the argument `**output` so I cannot allocate memory inside the method callback. Thats why I've assumed that open62541 allocates memory for me that I can simply use. But it yields a SEGFAULT...

Pro, 2018-08-22:
The `UA_Variant *output` should already be allocated for you, so theoretically your code should work.

Can you try to run the script with gdb to see where the segfault happens? And maybe also with valgrind, which gives even better error messages.

NIPE-SYSTEMS, 2018-08-22:
Thanks for the tips @Pro!

How can I help you with my Windows environment without gdb or valgrind? Are there any tricks for Visual Studio 2015 to achieve an equivalent debugging output?

(As a plan B, I'm trying to get it working on my Linux machine.)

Pro, 2018-08-22:
Then you can try to use Dr. Memory, which is something similar to valgrind.

See the appveyor scripts on how to install/use it:
https://github.com/open62541/open62541/tree/master/tools/appveyor

We are using it for memory leak testing on Windows and it should give more info if you are trying to access an invalid memory address.

Additionally, what also helps (us and you) if you can post a MCVE (https://stackoverflow.com/help/mcve), i.e. a full program as short as possible which shows your problem.
With that it's easier for us to see what you are doing, and most of the time it helps you to pinpoint the issue yourself if it isn't a bug. If it is a bug it helps us to easily reproduce it and fix it.

NIPE-SYSTEMS, 2018-08-23:
Thanks @Pro for the tips!

That's weird: Now I even cannot reproduce on Windows. Seems that I've had a bug somewhere in my code. Unfortunately I don't know what triggered it. I'm closing this now and will reopen the issue if it occurs again.

Thanks for the support everyone!

closed 2018-08-23 by NIPE-SYSTEMS

fabian727, 2018-08-23:
I do not have any method with output arguments. Though I integrated this part of code into one method-call. I don' t get any Segfault or other problems.

```c
    outputSize = 2;
    output = new UA_Variant[outputSize];

    UA_Boolean booltmp = true;
    UA_String stringtmp = UA_String_fromChars("it's me");

    UA_Variant_setScalarCopy(&output[0],&booltmp,&UA_TYPES[UA_TYPES_BOOLEAN]);
    UA_Variant_setScalarCopy(&output[1],&stringtmp,&UA_TYPES[UA_TYPES_STRING]);
```