open62541/open62541 #2002: Windows CE 6.0

TechAspirant, 2018-08-22:
Hello,

Can anyone have compiled open62541.c & open62541.h for windows compact embedded 6.0 ?

HolgerJeromin, 2018-08-22:
Which compiler do you have in mind?

TechAspirant, 2018-08-22:
I am getting number of errors while building lie below
What is the cause for this error Is my open62541.h is not valid for that platform
> Error	1	error C2061: syntax error : identifier 'UA_Boolean'	d:\parth\projects\opc ua wince\opcuaserver\open62541.h	607	opcuaserver

TechAspirant, 2018-08-22:
Compiled Open62541 and created server in windows CE 6.0 Its working.

closed 2018-08-22 by TechAspirant

JesperVilladsen, 2019-03-18:
Hi,
Where can I find the source code you successfully compiled and does it include version Open62541 v0.3.0?
I am using WinCE and Visual Studio 2005. I am new to gitHub, thank you in advance
@TechAspirant