open62541/open62541 #2003: Events triggering twice

fabian727, 2018-08-22:
Hi @ all,

I have seen you made some progress with the Events, but still I have the issue, that my events are firing twice.

I updated to the latest release (25fd8bf0e4bdfa017ed941a3e35dc4f72f0878f7) and recompiled everything.

I created a method, which gets severity and message as input parameters, sets date and sourcename and triggers it. Here the code (pretty the same as your example):
```c
UA_StatusCode
genEventMethodCallback(UA_Server *server,
					   const UA_NodeId *sessionId, void *sessionHandle,
				       const UA_NodeId *methodId, void *methodContext,
					   const UA_NodeId *objectId, void *objectContext,
					   size_t inputSize, const UA_Variant *input,
					   size_t outputSize, UA_Variant *output) {
	UA_StatusCode 	  retval;
	UA_LocalizedText *message;
	UA_UInt16 		 *severity;
	UA_NodeId  	      eventType = UA_NODEID_NUMERIC(2,UA_NS2ID_HMCTESTERROREVENTTYPE);
    UA_NodeId 		  eventNodeId; //this will buffer the NodeId given by the server
    UA_NodeId		  originId = UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER);
	//go through every input value.
	//If there are not enough inputs made, fill them with default values
	//so the input-array has to be filled bottom up. There can't be a value jumped over
    if(inputSize >= 1) {
		severity = (UA_UInt16*) input[0].data;
	} else {
		severity  = new UA_UInt16();
		*severity = 100;
	}
	if(inputSize >= 2) {
		message = (UA_LocalizedText*) input[1].data;
	} else {
		message = UA_LocalizedText_new();
		*message = UA_LOCALIZEDTEXT_ALLOC(LOCAL,"no Text");
	}

	//sets EventType
    retval = UA_Server_createEvent(server, eventType, &eventNodeId);

	//only go ahead, if all previous actions were good
	//those are all mandatory properties
	if(retval == UA_STATUSCODE_GOOD) {
		UA_QualifiedName originName;
		UA_Server_readBrowseName(server,originId,&originName);
		retval |= UA_Server_writeObjectProperty_scalar(server, eventNodeId, UA_QUALIFIEDNAME(0,(char*) "SourceName"),
													   &originName.name, &UA_TYPES[UA_TYPES_STRING]);
	}
	if(retval == UA_STATUSCODE_GOOD) {
		UA_UtcTime time = UA_DateTime_now();	//stamp as near as possible to the event
		retval |= UA_Server_writeObjectProperty_scalar(server, eventNodeId, UA_QUALIFIEDNAME(0,(char*) "Time"),
													   &time, &UA_TYPES[UA_TYPES_UTCTIME]);
	}
	if(retval == UA_STATUSCODE_GOOD) {
		retval |= UA_Server_writeObjectProperty_scalar(server, eventNodeId, UA_QUALIFIEDNAME(0, (char*)"Message"),
													   message, &UA_TYPES[UA_TYPES_LOCALIZEDTEXT]);
	}
	if(retval == UA_STATUSCODE_GOOD) {
		retval |= UA_Server_writeObjectProperty_scalar(server, eventNodeId, UA_QUALIFIEDNAME(0,(char*) "Severity"),
													   severity, &UA_TYPES[UA_TYPES_UINT16]);
	}
	//the optional PropertyType "LocalTime" isn't set

	if(retval == UA_STATUSCODE_GOOD) {
		//sets the mandatory fields: SourceNode, ReceiveTime, EventId
		retval |= UA_Server_triggerEvent(server, eventNodeId,
										 originId, NULL, UA_TRUE);
	}
    return retval;
}
```

The first PublishResponse contains at the moment this:
```
NotificationMessage: NotificationMessage
    SequenceNumber: 1
    PublishTime: Aug 22, 2018 12:24:52.206726000 CEST
    NotificationData: Array of ExtensionObject
        ArraySize: 1
        [0]: ExtensionObject
            TypeId: ExpandedNodeId
            EncodingMask: 0x01, has binary body
            EventNotificationList: EventNotificationList
                Events: Array of EventFieldList
                    ArraySize: 1
                    [0]: EventFieldList
                        ClientHandle: 0
                        EventFields: Array of Variant
                            ArraySize: 1
                            [0]: Variant
                                Variant Type: Null (0x00)
```
The second PublishResponse
```
NotificationMessage: NotificationMessage
    SequenceNumber: 2
    PublishTime: Aug 22, 2018 12:24:52.431619000 CEST
    NotificationData: Array of ExtensionObject
        ArraySize: 1
        [0]: ExtensionObject
            TypeId: ExpandedNodeId
            EncodingMask: 0x01, has binary body
            EventNotificationList: EventNotificationList
                Events: Array of EventFieldList
                    ArraySize: 1
                    [0]: EventFieldList
                        ClientHandle: 1
                        EventFields: Array of Variant
                            ArraySize: 2
                            [0]: Variant
                                Variant Type: LocalizedText (0x15)
                                Value: LocalizedText
                            [1]: Variant
                                Variant Type: UInt16 (0x05)
                                UInt16: 100
```

Another function, pretty the same as above, just other message-text shows it a bit better:

```
Events: Array of EventFieldList
    ArraySize: 2
    [0]: EventFieldList
        ClientHandle: 2
        EventFields: Array of Variant
            ArraySize: 2
            [0]: Variant
                Variant Type: LocalizedText (0x15)
                Value: LocalizedText
            [1]: Variant
                Variant Type: UInt16 (0x05)
                UInt16: 100
    [1]: EventFieldList
        ClientHandle: 6
        EventFields: Array of Variant
            ArraySize: 2
            [0]: Variant
                Variant Type: LocalizedText (0x15)
                Value: LocalizedText
            [1]: Variant
                Variant Type: UInt16 (0x05)
                UInt16: 100
```
There is one trigger, but two events which are identical, but do not hold all data.

I allready asked this question here: https://groups.google.com/forum/#!topic/open62541/Y3gQjQUj7Wg
but got referenced for asking in the git-repo itself.

So my 2. question would be: why do you configure a second namespace (urn:unconfigured:application), which is not overwritten if I add one. I just can rename it. But there is no Node/object/type... referencing it.

Thanks in Advance,

Fabian

Pro, 2018-08-22:
@aDogCalledSpot can you have a look at the first issue

@fabian727 please do not post two questions in the same issue. This should be a separate thread.

Pro, 2018-08-22:
It looks like the events are not identical. Can you check what the actual difference is between those two?
Also can you add additional output in the genEvent Method to make sure that you are only calling it once

fabian727, 2018-08-23:
@aDogCalledSpot  @Pro I just added a print(__PRETTY_FUNCTION__) 

```
...
[2018-08-23 08:52:01.504 (UTC+0200)] trace/network      Connection 8 | Activity on the socket
[2018-08-23 08:52:01.504 (UTC+0200)] trace/network      Connection 8 | Received a packet.
[2018-08-23 08:52:01.504 (UTC+0200)] trace/channel      Connection 8 | SecureChannel 3 | Process a MSG
[2018-08-23 08:52:01.504 (UTC+0200)] debug/session      Connection 8 | SecureChannel 3 | Session ns=1;g=c524343f-618c-3639-8538-533548747707 | Processing ReadRequest
[2018-08-23 08:52:01.504 (UTC+0200)] debug/session      Connection 8 | SecureChannel 3 | Session ns=1;g=c524343f-618c-3639-8538-533548747707 | Read the attribute 13
[2018-08-23 08:52:01.572 (UTC+0200)] debug/session      Connection 8 | SecureChannel 3 | Session ns=1;g=c524343f-618c-3639-8538-533548747707 | Subscription 1 | Publish Callback
[2018-08-23 08:52:01.672 (UTC+0200)] debug/session      Connection 8 | SecureChannel 3 | Session ns=1;g=c524343f-618c-3639-8538-533548747707 | Subscription 1 | Publish Callback
[2018-08-23 08:52:01.772 (UTC+0200)] debug/session      Connection 8 | SecureChannel 3 | Session ns=1;g=c524343f-618c-3639-8538-533548747707 | Subscription 1 | Publish Callback
[2018-08-23 08:52:01.872 (UTC+0200)] debug/session      Connection 8 | SecureChannel 3 | Session ns=1;g=c524343f-618c-3639-8538-533548747707 | Subscription 1 | Publish Callback
[2018-08-23 08:52:01.905 (UTC+0200)] trace/network      Connection 8 | Activity on the socket
[2018-08-23 08:52:01.905 (UTC+0200)] trace/network      Connection 8 | Received a packet.
[2018-08-23 08:52:01.905 (UTC+0200)] trace/channel      Connection 8 | SecureChannel 3 | Process a MSG
[2018-08-23 08:52:01.905 (UTC+0200)] debug/session      Connection 8 | SecureChannel 3 | Session ns=1;g=c524343f-618c-3639-8538-533548747707 | Processing CallRequest

 UA_StatusCode genEventMethodCallback(UA_Server*, const UA_NodeId*, void*, const UA_NodeId*, void*, const UA_NodeId*, void*, size_t, const UA_Variant*, size_t, UA_Variant*)

[2018-08-23 08:52:01.906 (UTC+0200)] debug/session      Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | Read the attribute 3
[2018-08-23 08:52:01.972 (UTC+0200)] debug/session      Connection 8 | SecureChannel 3 | Session ns=1;g=c524343f-618c-3639-8538-533548747707 | Subscription 1 | Publish Callback
[2018-08-23 08:52:01.972 (UTC+0200)] debug/session      Connection 8 | SecureChannel 3 | Session ns=1;g=c524343f-618c-3639-8538-533548747707 | Subscription 1 | Sending out a publish response with 1 notifications
[2018-08-23 08:52:01.993 (UTC+0200)] trace/network      Connection 8 | Activity on the socket
[2018-08-23 08:52:01.993 (UTC+0200)] trace/network      Connection 8 | Received a packet.
[2018-08-23 08:52:01.993 (UTC+0200)] trace/channel      Connection 8 | SecureChannel 3 | Process a MSG
[2018-08-23 08:52:01.993 (UTC+0200)] debug/session      Connection 8 | SecureChannel 3 | Session ns=1;g=c524343f-618c-3639-8538-533548747707 | Processing PublishRequest
[2018-08-23 08:52:01.993 (UTC+0200)] debug/session      Connection 8 | SecureChannel 3 | Session ns=1;g=c524343f-618c-3639-8538-533548747707 | Queued a publication message
[2018-08-23 08:52:02.072 (UTC+0200)] debug/session      Connection 8 | SecureChannel 3 | Session ns=1;g=c524343f-618c-3639-8538-533548747707 | Subscription 1 | Publish Callback
[2018-08-23 08:52:02.171 (UTC+0200)] debug/session      Connection 8 | SecureChannel 3 | Session ns=1;g=c524343f-618c-3639-8538-533548747707 | Subscription 1 | Publish Callback
[2018-08-23 08:52:02.272 (UTC+0200)] debug/session      Connection 8 | SecureChannel 3 | Session ns=1;g=c524343f-618c-3639-8538-533548747707 | Subscription 1 | Publish Callback
...
```
It' s a bit of output, thanks to debugging, but there is only once the name of my method. So I am a 100% sure, it is only called once.

One difference is: the genEvent (twice executed) is defined in the xml-namespace-file as method in the object-directory. There is no Type of it.
I add the callback with this lines
```
typedef struct MethodCallbacks{
	UA_MethodCallback callback;
	const char *methodName;
	const char *parentName;
} MethodCallbacks_t;

static MethodCallbacks_t combiM[] = {
		{genEventMethodCallback,	"GenerateEvent","Objects"},
...
};

void addfunctions() {
...
	retval  = getMethodTypeId(server, ns, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), combiM[0].methodName, methodId);
	retval |= UA_Server_setMethodNode_callback(server,methodId,combiM[0].callback);
```

My other methods are components of an objectType which is generated as object during startup of the server. The callbacks are added to the Method of the objectTypes as methods are only referencing. 

Here is the xml-namespace-part of my events (I had to change their name etc (replaced with FOO), but there are no more references or else):
```
    <UAObjectType IsAbstract="false" NodeId="ns=1;i=1002" BrowseName="1:FOO">
        <DisplayName>FOO</DisplayName>
        <Description>FOO</Description>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=2041</Reference>
        </References>
    </UAObjectType>
    <UAObjectType IsAbstract="false" NodeId="ns=1;i=1018" BrowseName="1:FOO2">
        <DisplayName>FOO2</DisplayName>
        <Description>FOO2</Description>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=2041</Reference>
        </References>
    </UAObjectType>
```

aDogCalledSpot, 2018-08-30:
What are you using to receive the events? It looks like the same event being triggered for two separate MonitoredItems, one of which has only one item in its filter and your event does not have this item.

If you are using UAExpert, then this is because UAExpert always creates a MonitoredItem which has a single item in its filter called "Changes", if your event does not have a "Changes" field, this is what you would expect to see.

aDogCalledSpot, 2018-09-03:
With "field" I mean a "property/attribute" of an event (i.e. EventType, SourceNode, etc.).

In the specification in Part 4 7.17.3 it says

> A null value is returned in the corresponding event field in the Publish response if the selected field is not part of the Event

This is why most of your second Publish response is empty. The first one is empty for the same reason but on a different MonitoredItem, not the one you are creating. UAExpert simply creates this one and only requests that a "Changes" field is filled - which you don't have, so a null value is left in its place.

fabian727, 2018-09-03:
so took me some time, but I think i got some of it (i hope so)

Why two events are triggered and steps to reproduce:

1 start the example server "tutorial_server_events", start wireshark
2 connect the UAExpert (don't subscribe to anything, just simply connect and nothing more!)
3 call the method to generate the event
4 look into the wireshark log, you will see in one of the PublishResponses after the methodCall an "eventNotification" with 1 EventField, which is empty

the open62541 server creates an event, though it wasn't subscribed. It is also out of time-context (default intervall of my UAExpert is 5.0seconds)

In my appended log you can see the default PublishResponse at 15 seconds. The next expected would be at 20seconds. Though there is one at 17.124, direct after the CallResponse, which has the EventNotificationMessageList
[EventWithoutSubscription.txt](https://github.com/open62541/open62541/files/2345462/EventWithoutSubscription.txt)
How can I upload WireShark files direct? 
[EventWithoutRegistration.zip](https://github.com/open62541/open62541/files/2345467/EventWithoutRegistration.zip)

fabian727, 2018-10-01:
@jpfr I am sorry,  Issue remains. I updated my repo and built everything new. Exact same behaviour.

In Wireshark I can see an empty package, without any manual subscription (UA Expert does its default subscription). Here again the Wireshark-log of connecting UA Expert, calling GenerateEvent and disconnecting.

[event_triggers_twice.pcapng.tar.gz](https://github.com/open62541/open62541/files/2433503/event_triggers_twice.pcapng.tar.gz)

aDogCalledSpot, 2018-10-15:
> I can see an empty package, without any manual subscription (UA Expert does its default subscription)

So there should be a package being sent, the one issued because of the default subscription and monitored item, which is the package you are seeing. In the log you sent you can also see the filter being passed with the monitored item and that none of these fields are covered by the GenerateEvent method.

fabian727, 2018-10-26:
you are right.

closed 2018-10-26 by fabian727