open62541/open62541 #2004: How to get value from client on OnWrite Callback in Server ?

TechAspirant, 2018-08-22:
Hello,

How can I get value from client OnWrite Callback of server so that I can force that value into server Node ?

NIPE-SYSTEMS, 2018-08-22:
Do you mean value callbacks on the server side, that are triggered when the client reads/writes a value? If yes, this might help: https://open62541.org/doc/current/tutorial_server_datasource.html

TechAspirant, 2018-08-22:
can I print value on OnWrite callback of the node ? means Information in const UA_Variant *data

```
static void onWrite(void *h, const UA_NodeId nodeid, const UA_Variant *data, const UA_NumericRange *range) {
    UA_LOG_INFO(logger, UA_LOGCATEGORY_USERLAND, "onWrite; handle: %i", (uintptr_t)h);
}
```

NIPE-SYSTEMS, 2018-08-22:
Yes you can with the help of e.g. casting.

The UA_Variant-typedef:

```cpp
typedef struct {
    const UA_DataType *type;      /* The data type description */
    UA_VariantStorageType storageType;
    size_t arrayLength;           /* The number of elements in the data array */
    void *data;                   /* Points to the scalar or array data */
    size_t arrayDimensionsSize;   /* The number of dimensions */
    UA_UInt32 *arrayDimensions;   /* The length of each dimension */
} UA_Variant;
```

Test for correct type stored in variant and receive the value:

```cpp
if(UA_Variant_hasScalarType(data, &UA_TYPES[UA_TYPES_INT16])) { // if you are paranoid you may also want to check the value rank and dimensions
    int16_t value = *(int16_t *)data->data;
}
```

In your case:

```cpp
static void onWrite(void *h, const UA_NodeId nodeid, const UA_Variant *data, const UA_NumericRange *range) {
    if(UA_Variant_hasScalarType(data, &UA_TYPES[UA_TYPES_INT16])) {
        UA_LOG_INFO(logger, UA_LOGCATEGORY_USERLAND, "onWrite; handle: %i, value: %i", (uintptr_t)h, *(int16_t *)data->data);
    }
}
```

Hope it helps.

TechAspirant, 2018-08-23:
Its working But When I implement same thing in OnRead Callback application gets crashed. Is there something wrong in my code ?

```
static void onRead(void *handle, const UA_NodeId nodeid, const UA_Variant *data, const UA_NumericRange *range) {
    UA_LOG_INFO(logger, UA_LOGCATEGORY_USERLAND, "onWrite; handle: %i, value: %i", (uintptr_t)handle, *(int16_t *)data->data);
}
```

NIPE-SYSTEMS, 2018-08-23:
I'm guessing the context of your question. Assuming you write server code and define data source callbacks for a value. Then there exist two callback types: a read callback and a write callback. One of them gets called when e.g. the client reads a value and the other gets called when e.g. the client writes to the value. This means that one callback informs you (i.e. the server) about a happened write and for the other you are responsible for providing a value that e.g. gets sent to the client (so the client reads the value).

Which means in your case that you cannot print the value because there is no value passed to your callback. Instead you need to construct a value. Again, please consider reading https://open62541.org/doc/current/tutorial_server_datasource.html and https://open62541.org/doc/current/tutorial_datatypes.html#variants for examples of callbacks and the concepts of variants.

If you want to write to the variant, your code might look like this:

```cpp
static void onRead(void *handle, const UA_NodeId nodeid, const UA_Variant *data, const UA_NumericRange *range) {
    int16_t value = 42;
    UA_Variant_setScalarCopy(data, &value, &UA_TYPES[UA_TYPES_INT16]);
}
```

If you have more questions, please provide more informations about the context:

* How are you using the callbacks: How do you make open62541 executing your callbacks? Some call like `hey_open62541_please_call_this_function(&onRead)`).
* Are you writing server code or client code?

TechAspirant, 2018-08-23:
Hello,

Thanks for nice explanation, I am writing server code.Information you have provided is useful you are right I can not print value on read callback unless i pass value to client on read call back.

regarding security channel can I use client certificate and verify at server side whether to allow connection or not ?
Else do I have to load server certificate at server side in order to establish secure channel between client and server during connection.

```
UA_ServerConfig config = UA_ServerConfig_standard;
config.serverCertificate = loadFile("server_cert.der");
```