open62541/open62541 #2005: nodeset on the myNS example produces error

kleskjr, 2018-08-22:
runing
`$ python ./nodeset_compiler.py --types-array=UA_TYPES --existing ../../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml --xml myNS.xml myNS`

produces myNS.c/h files as expected, but then compiling them into a server fails.

```
gcc -std=c99 open62541.c myNS.c open6_server.c -o open6_server 

In file included from myNS.c:4:0:
myNS.h:13:1: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘extern’
 extern UA_StatusCode myNS(UA_Server *server);
...
```

The myNS.h file looks like:
```
/* WARNING: This is a generated file.
 * Any manual changes will be overwritten. */

#ifndef MYNS_H_
#define MYNS_H_

#include "open62541.h"

_UA_BEGIN_DECLS

extern UA_StatusCode myNS(UA_Server *server);

_UA_END_DECLS

#endif /* MYNS_H_ */
```
Removing the 2 lines containng _UA_BEGIN_DECLS and _UA_END_DECLS solves the issue. So it looks like that the problem is in missing definitions in the open62541.c file

Pro, 2019-01-03:
Can you post the content of your myNS.xml?

Also, which version are you using (commit id) and which operating system, including the compiler version.

kleskjr, 2019-01-09:
This is the example myNS.xml from the docs (at the end of the message). The verison that gives me this issue is from November with hash 5a06ca2471fa973cddbac90553cd516db0809f04. Interestingly, an older version from June (9c9615b68c629b02c216440367847c6dc8b21104) results in the following header (did not check whether it compiles):

```
/* WARNING: This is a generated file.
 * Any manual changes will be overwritten. */

#ifndef MYNS_H_
#define MYNS_H_

#include "open62541.h"

#ifdef __cplusplus
extern "C" {
#endif

extern UA_StatusCode myNS(UA_Server *server);

#ifdef __cplusplus
}
#endif

#endif /* MYNS_H_ */

```

myNS.xml:
```
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
           xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd"
           xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd"
           xmlns:s1="http://yourorganisation.org/example_nodeset/"
           xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://yourorganisation.org/example_nodeset/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="Boolean">i=1</Alias>
        <Alias Alias="UInt32">i=7</Alias>
        <Alias Alias="String">i=12</Alias>
        <Alias Alias="HasModellingRule">i=37</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="HasSubtype">i=45</Alias>
        <Alias Alias="HasProperty">i=46</Alias>
        <Alias Alias="HasComponent">i=47</Alias>
        <Alias Alias="Argument">i=296</Alias>
    </Aliases>
    <Extensions>
        <Extension>
            <ModelInfo Tool="UaModeler" Hash="Zs8w1AQI71W8P/GOk3k/xQ=="
                       Version="1.3.4"/>
        </Extension>
    </Extensions>
    <UAReferenceType NodeId="ns=1;i=4001" BrowseName="1:providesInputTo">
        <DisplayName>providesInputTo</DisplayName>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">
                i=33
            </Reference>
        </References>
        <InverseName Locale="en-US">inputProcidedBy</InverseName>
    </UAReferenceType>
    <UAObjectType IsAbstract="true" NodeId="ns=1;i=1001"
                  BrowseName="1:FieldDevice">
        <DisplayName>FieldDevice</DisplayName>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">
                i=58
            </Reference>
            <Reference ReferenceType="HasComponent">ns=1;i=6001</Reference>
            <Reference ReferenceType="HasComponent">ns=1;i=6002</Reference>
[... 124 lines cut ...]
        </References>
        <Value>
            <ListOfExtensionObject>
                <ExtensionObject>
                    <TypeId>
                        <Identifier>i=297</Identifier>
                    </TypeId>
                    <Body>
                        <Argument>
                            <Name>stopped</Name>
                            <DataType>
                                <Identifier>i=1</Identifier>
                            </DataType>
                            <ValueRank>-1</ValueRank>
                            <ArrayDimensions></ArrayDimensions>
                            <Description/>
                        </Argument>
                    </Body>
                </ExtensionObject>
            </ListOfExtensionObject>
        </Value>
    </UAVariable>
</UANodeSet>

```