open62541/open62541 #2007: Windows Compact Embedded

TechAspirant, 2018-08-23:
Hello,

Anyone have compiled open62541.c & open62541.h latest release 0.3rc2 for windows compact embedded 6.0 , 7.0 or 2013 ?

crafti5, 2018-08-23:
Yes I have done this, but I haven't time to publish this as a pull request. But I can try to upload the amalgated version. If someone can apply this to current master brnach?

TechAspirant, 2018-08-23:
Hello,
can apply this to current master branch? How can I as I am new to github ?

TechAspirant, 2018-08-29:
@crafti5 Please tell me how can I make request for your windows embedded solution ?

crafti5, 2018-08-29:
@TechAspirant I dont know what you mean, I have changed amalgated files and there is a need to get this as a new arch.

crafti5, 2018-08-29:
I have created a pull request: https://github.com/open62541/open62541/pull/2026

jpfr, 2018-11-18:
This was added with #2059.

closed 2018-11-18 by jpfr