open62541/open62541 #2009: ServerCapabilities

fabian727, 2018-08-23:
Part 4 of the specification (chapter 6.3.2) states that every server could have a capability. Why did you pack it into the Discovery-Part? I think it should be generic, so in the default configuration

[from here](https://github.com/open62541/open62541/blob/master/include/ua_server_config.h#L74)
```c
#ifdef UA_ENABLE_DISCOVERY
    UA_String mdnsServerName;
    size_t serverCapabilitiesSize;
    UA_String *serverCapabilities;
#endif
```

so more like this:
```c
#ifdef UA_ENABLE_DISCOVERY
    UA_String mdnsServerName;
#endif
    size_t serverCapabilitiesSize;
    UA_String *serverCapabilities;
```

fabian727, 2018-10-24:
What is about the ServerCapabilites?

Pro, 2018-11-06:
The value of server capabilities is only read and used for the multicast discovery functionality.
As long as discovery is disabled, the server capability is not needed and that is why it is correctly within the ifdef.

closed 2018-11-06 by Pro