open62541/open62541 #2010: Matlab client cannot connect to open62541 opcua server

braiankonzgen, 2018-08-23:
Hi everyone, 
I am trying to connect a matlab toolbox opc ua client on an open62541 server but with no success.

I can find the server and connect to a client using the UAexpert API.

But when I try to connect the client in matlab i get the following error.

In Matlab I run:
localServers = opc.ua.getServerInfo('LIT31')
Warning: Some endpoints could not be retreived:
	open62541-based OPC UA Application: The hostname could
    not be resolved. 
> In opc.ua.getServerInfo (line 66) 

localServers = 

OPC UA ServerInfo 'open62541-based OPC UA Application':

   Connection Information
    Hostname: 'LIT31'
        Port: 4840

>>  uaClient = opcua(localServers(1))

uaClient = 

OPC UA Client open62541-based OPC UA Application:
                     Hostname: LIT31
                         Port: 4840
                      Timeout: 10

                       Status: Disconnected

Client functions
>> connect(uaClient)
Error using opc.ua.Client/connect (line 420)
Could not retrieve endpoints for server. Received message:
The hostname could not be resolved.

Best regards

jpfr, 2018-08-24:
> The hostname could not be resolved.

Is the hostname reachable in general? I.e. can you `ping hostname`?

braiankonzgen, 2018-08-24:
dear Julius Pfrommer,
yes I can connect it to the OPC server provided as an example by the OPC foundation, I can also ping the OPC server 62541, but the matlab client can not connect.

Pro, 2018-08-24:
This looks like a misconfiguration somewhere.

See the output:

```
Warning: Some endpoints could not be retreived:
open62541-based OPC UA Application: The hostname could
not be resolved.
```

This looks to mee that you set `open62541-based OPC UA Application` as the endpoint, but it should be smth like `opc.tcp://localhost:4840`

braiankonzgen, 2018-08-27:
My matlab UA client can not find server url endpoint
[image]

braiankonzgen, 2018-08-30:
I can even edit, and manually set the EndpointUrl (as suggested). But when I try to reconnect the client it returns the following errors
>> connect(uaClient)
Error using asyncioimpl.Channel/init
Endpoint Url must be a non-empty string.

Error in asyncio.Channel (line 114)
            customProps = obj.ChannelImpl.init(options);

Error in opc.ua.Channel (line 19)
            obj@asyncio.Channel(deviceName, converterName,
            varargin{:});

Error in opc.ua.Client/connect (line 472)
                        clnt.AsyncChannel = opc.ua.Channel(options,
                        [Inf,0]);
 
>>

and the EndpointUrl field is empty again
[image]

tks

jvledesma, 2018-09-21:
I am experiencing the same issues with this toolbox.
I have also tried to change the EndpointUrl manually as jpfr suggests.
[image]
[image]
¨
I still get the same issue:
```
Error using opc.ua.Client/connect (line 499)
The hostname could not be resolved.
```
I successfully connect to this server with other clients like python or UAExpert. but not with matlab nor simulink

Appreciated any news/hints on this topic

Best Regards

matkonnerth, 2019-01-22:
Hi, I had the same issue, now managed to connect when I provide manually the hostname to IP-address resolution in system32/Drivers/etc/hosts.
open62541 is running in a vm, hostname "debian-9", ip: 192.168.110.42

**not working**
When I use the IP-address in Matlab, and have no entry in system32/Drivers/etc/hosts the EndpointUrl is empty:

[image]

**working**
When I use the hostname or IP-address (I entered it manually in system32/Drivers/etc/hosts), it's working:
in Matlab: _client = opcua('debian-9', 4840)_

[image]

The EndpointUrl ist now set to a valid string.

Tested with Matlab R2018b, win64

teccheck, 2025-04-15:
It seems to be solved and there's been no more activity