open62541/open62541 #2028: Feature Request: Reverse Hello for Firewall Traversal

jpfr, 2018-08-30:
Part 6 (since v1.04) defines a Reverse Hello (RHE) message to make the initial handshake from the server to the client. We don't support this atm.

It would probably require some changes to our networking layer as well.

Pro, 2019-04-12:
Related PR for network layer rework: #2408

jopi2016, 2021-11-16:
Hi,
I'm also interested in the Reverse Hello Feature and went trough the different network layer rework pull requests. Although it is not very clear to me about how's work going with the different Pull Requests and when they are finished so that the Reverse Hello Feature could step forward.
Could you clarify this a bit?

Would it be apart from the network layer rework possible to implement the Reverse Hello Feature myself or is this not possible with the current code base?

Thanks in advance

RavilN, 2022-02-28:
Hi,
I wonder, is anybody working on this feature? Are there any plans?
Thank you, Ravil

teccheck, 2025-04-15:
This is implemented now

closed 2025-04-15 by teccheck