open62541/open62541 #2032: What is correct way for defining customised enumeration?

crafti5, 2018-08-31:
hi all,

I'm struggling of the usage for creating self defined Enumerations, the example for custom data type isn't helpfully for me. If we have an enumeration like 
Turbines:
0 = Off
1 = On
2 = Maintenance

What is correct way to implement this and make this known to the stack?
An example would be great for enumeration, I would write one if I got this working :)

Best regards

Notes: I've tried the nodeset_compiler but with this new enumeration I got a message due "identifier not found".

jpfr, 2018-09-02:
Datatypes are not generated by the nodeset compiler, but by /tools/generate_datatypes.py.

The AppicationType enum is defined here: https://github.com/open62541/open62541/blob/master/tools/schema/Opc.Ua.Types.bsd#L1126

And generates the following code:

```c
/**                                                                                                                                                                                                                  
 * ApplicationType                                                                                                                                                                                                   
 * ^^^^^^^^^^^^^^^                                                                                                                                                                                                   
 * The types of applications. */
typedef enum {
    UA_APPLICATIONTYPE_SERVER = 0,
    UA_APPLICATIONTYPE_CLIENT = 1,
    UA_APPLICATIONTYPE_CLIENTANDSERVER = 2,
    UA_APPLICATIONTYPE_DISCOVERYSERVER = 3,
    __UA_APPLICATIONTYPE_FORCE32BIT = 0x7fffffff
} UA_ApplicationType;
UA_STATIC_ASSERT(sizeof(UA_ApplicationType) == sizeof(UA_Int32), enum_must_be_32bit);
```

An additional representation in the information model can be generated.
But the (C-language) type generation and the nodeset compiler are kinda independent for the time being.

>  An example would be great for enumeration, I would write one if I got this working :)

Yeah!

starki93, 2018-09-06:
Hey all, i have a answer for this problem, but at the same time a question.

For testing my server i'm using the UaExpert client.

I defined my custom enum like this: 
`/**
 * TestEnum
 * ^^^^^^^^^^^^^^^^^
 */
typedef enum {
    TestEnum_NONE = 0,
    TestEnum_ONE = 1,
    TestEnum_TWO = 2,
    __TestEnum_FORCE32BIT = 0x7fffffff
} TestEnum;
UA_STATIC_ASSERT(sizeof(TestEnum) == sizeof(UA_Int32), enum_must_be_32bit);

static UA_DataTypeMember TestEnum_members[1] = {
{
    UA_TYPENAME("") /* .memberName */
    UA_TYPES_INT32, /* .memberTypeIndex */
    0, /* .padding */
    true, /* .namespaceZero */
    false /* .isArray */
}};

static const UA_DataType TestEnumType = {
        UA_TYPENAME("TestEnum")             /* .typeName */
        {1, UA_NODEIDTYPE_NUMERIC, {4242}}, /* .typeId */
        sizeof(TestEnum_members),                   /* .memSize */
        0,                               /* .typeIndex, in the array of custom types */
        1,                               /* .membersSize */
        false,                           /* .builtin */
        true,                            /* .pointerFree */
        false,                           /* .overlayable (depends on endianness and
                                         the absence of padding) */
        0,                               /* .binaryEncodingId, the numeric
                                         identifier used on the wire (the
                                         namespaceindex is from .typeId) */
        TestEnum_members
};`

and add the DataType like this after add all members to the stack:

`static void
addTestEnumType(UA_Server *server) {

    // add datatype node
    UA_DataTypeAttributes ddaatt = UA_DataTypeAttributes_default;
    ddaatt.description = UA_LOCALIZEDTEXT(const_cast<char*>("en-US"), const_cast<char*>("TestEnumDataType"));
    ddaatt.displayName = UA_LOCALIZEDTEXT(const_cast<char*>("en-US"), const_cast<char*>("TestEnumDataType"));

    UA_Server_addDataTypeNode(server, TestEnumType.typeId,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_ENUMERATION),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
                              UA_QUALIFIEDNAME(1, const_cast<char*>("TestEnumDataType")),
                              ddaatt, nullptr, nullptr);

    // add enumstrings property
    UA_VariableAttributes pattr = UA_VariableAttributes_default;
    pattr.description = UA_LOCALIZEDTEXT(const_cast<char*>("en-US"), const_cast<char*>("EnumStrings"));
    pattr.displayName = UA_LOCALIZEDTEXT(const_cast<char*>("en-US"), const_cast<char*>("EnumStrings"));
    pattr.dataType = UA_TYPES[UA_TYPES_LOCALIZEDTEXT].typeId;

    UA_UInt32 axesDeminsion[1] = {0};
    pattr.valueRank = 1;                               // array demision rank 1 = array
    pattr.arrayDimensionsSize = 1;                     // array deminsion size  1 = array
    pattr.arrayDimensions = axesDeminsion;             // array deminsions -> array of size [ number of axes ]

    UA_LocalizedText* test2 = new UA_LocalizedText[3];
    test2[0] = UA_LOCALIZEDTEXT(const_cast<char*>(""), const_cast<char*>("NONE"));
    test2[1] = UA_LOCALIZEDTEXT(const_cast<char*>(""), const_cast<char*>("ONE"));
    test2[2] = UA_LOCALIZEDTEXT(const_cast<char*>(""), const_cast<char*>("TWO"));

    UA_Variant_setArray(&pattr.value, test2, 3, &UA_TYPES[UA_TYPES_LOCALIZEDTEXT]);

    UA_Server_addVariableNode(server, UA_NODEID_NUMERIC(1, 4001),
                              TestEnumType.typeId, UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
                              UA_QUALIFIEDNAME(0, const_cast<char*>("EnumStrings")),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_PROPERTYTYPE), pattr, nullptr, nullptr);

}`

and i add variable like this:

`    // TEST ENUM
    TestEnum p;
    p = TestEnum_TWO;
    UA_VariableAttributes vattr = UA_VariableAttributes_default;

    vattr.description = UA_LOCALIZEDTEXT(const_cast<char*>("en-US"), const_cast<char*>("TestEnum"));
    vattr.displayName = UA_LOCALIZEDTEXT(const_cast<char*>("en-US"), const_cast<char*>("TestEnum"));
    vattr.dataType = TestEnumType.typeId;
    vattr.valueRank = -1;
    vattr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
    UA_Variant_setScalar(&vattr.value, &p, &TestEnumType);

    UA_Server_addVariableNode(server, UA_NODEID_STRING(1, "TestEnum"),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                              UA_QUALIFIEDNAME(1, "TestEnum"),
                              UA_NODEID_NUMERIC(1, 4000), vattr, nullptr, nullptr);`

For some reason it's important that UA_QUALIFIEDNAME "EnumStrings" is at namespace index 0.

Is this the right way i have defined the enum? Is it possible  to send an integer that is higher as 3  from client and how would my server react to this?

crafti5, 2018-09-17:
@starki93 I have tried your code you posted but unfortunately I have the same response "UA_STATUSCODE_BADTYPEDEFINITIONINVALID"

@Pro & @jpfr Can you guys help me with the issue? The following shows where the problem comes from (Note: added WEC7 is my current latest commit).

` 
/* Get the node type. There must be a typedefinition for variables, objects
     * and type-nodes. See the above checks. */
    if(!UA_NodeId_isNull(typeDefinitionId)) {
        /* Get the type node */
        type = UA_Nodestore_get(server, typeDefinitionId);
        if(!type) {
            UA_LOG_NODEID_WRAP(typeDefinitionId, UA_LOG_INFO_SESSION(server->config.logger, session,
                                "AddNodes: Node type %.*s not found",
                                (int)nodeIdStr.length, nodeIdStr.data));
            retval = UA_STATUSCODE_BADTYPEDEFINITIONINVALID;
            goto cleanup;
        }
`

It seems that type is not correct or has missing entries / definitions?

crafti5, 2018-09-19:
I have wrote an example code based on the server tutorial and the enum from @starki93. Can anyone verify this please?

```c
#include <signal.h>
#include <stdio.h>
#include "open62541.h"

UA_Boolean running = true;

/**
 * TestEnum
 * ^^^^^^^^^^^^^^^^^
 */
typedef enum { TestEnum_NONE = 0, TestEnum_ONE = 1, TestEnum_TWO = 2, __TestEnum_FORCE32BIT = 0x7fffffff } TestEnum;
UA_STATIC_ASSERT(sizeof(TestEnum) == sizeof(UA_Int32), enum_must_be_32bit);

static UA_DataTypeMember TestEnum_members[1] = {{
    UA_TYPENAME("") /* .memberName */
    UA_TYPES_INT32, /* .memberTypeIndex */
    0,              /* .padding */
    true,           /* .namespaceZero */
    false           /* .isArray */
}};

static const UA_DataType TestEnumType = {UA_TYPENAME("TestEnum")             /* .typeName */
                                         {1, UA_NODEIDTYPE_NUMERIC, {4242}}, /* .typeId */
                                         sizeof(TestEnum_members),           /* .memSize */
                                         0,                                  /* .typeIndex, in the array of custom types */
                                         1,                                  /* .membersSize */
                                         false,                              /* .builtin */
                                         true,                               /* .pointerFree */
                                         false,                              /* .overlayable (depends on endianness and
                                                                             the absence of padding) */
                                         0,                                  /* .binaryEncodingId, the numeric
                                                                             identifier used on the wire (the
                                                                             namespaceindex is from .typeId) */
                                         TestEnum_members};

static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}

static void addTestEnumType(UA_Server* server) {
    // add datatype node
    UA_DataTypeAttributes ddaatt = UA_DataTypeAttributes_default;
    ddaatt.description = UA_LOCALIZEDTEXT(const_cast<char*>("en-US"), const_cast<char*>("TestEnumDataType"));
    ddaatt.displayName = UA_LOCALIZEDTEXT(const_cast<char*>("en-US"), const_cast<char*>("TestEnumDataType"));

    UA_Server_addDataTypeNode(server, TestEnumType.typeId, UA_NODEID_NUMERIC(0, UA_NS0ID_ENUMERATION), UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
                              UA_QUALIFIEDNAME(1, const_cast<char*>("TestEnumDataType")), ddaatt, NULL, NULL);

    // add enumstrings property
    UA_VariableAttributes pattr = UA_VariableAttributes_default;
    pattr.description = UA_LOCALIZEDTEXT(const_cast<char*>("en-US"), const_cast<char*>("EnumStrings"));
    pattr.displayName = UA_LOCALIZEDTEXT(const_cast<char*>("en-US"), const_cast<char*>("EnumStrings"));
    pattr.dataType = UA_TYPES[UA_TYPES_LOCALIZEDTEXT].typeId;

    UA_UInt32 axesDeminsion[1] = {0};
    pattr.valueRank = 1;                    // array demision rank 1 = array
    pattr.arrayDimensionsSize = 1;          // array deminsion size  1 = array
    pattr.arrayDimensions = axesDeminsion;  // array deminsions -> array of size [ number of axes ]

    UA_LocalizedText* test2 = new UA_LocalizedText[3];
    test2[0] = UA_LOCALIZEDTEXT(const_cast<char*>(""), const_cast<char*>("NONE"));
    test2[1] = UA_LOCALIZEDTEXT(const_cast<char*>(""), const_cast<char*>("ONE"));
    test2[2] = UA_LOCALIZEDTEXT(const_cast<char*>(""), const_cast<char*>("TWO"));

    UA_Variant_setArray(&pattr.value, test2, 3, &UA_TYPES[UA_TYPES_LOCALIZEDTEXT]);

[... 58 lines cut ...]
    UA_Server_write(server, &wv);

    /* Reset the variable to a good statuscode with a value */
    wv.value.hasStatus = false;
    wv.value.value = myVar;
    wv.value.hasValue = true;
    UA_Server_write(server, &wv);
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_ServerConfig* config = UA_ServerConfig_new_default();
    UA_Server* server = UA_Server_new(config);
    addTestEnumType(server);
    addVariable(server);
    writeVariable(server);

    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return (int)retval;
}
``

crafti5, 2018-09-25:
it seems to me that the function "findOccupiedSlot" does not find the required Node.

`
do {
        entry = ns->entries[(UA_UInt32)idx];
        if(entry > UA_NODEMAP_TOMBSTONE &&
           UA_NodeId_equal(&entry->node.nodeId, nodeid))
            return &ns->entries[(UA_UInt32)idx];
        idx += hash2;
        if(idx >= size)
            idx -= size;
    } while((UA_UInt32)idx != startIdx && entry);
`

AlexSTI, 2018-10-10:
Check the proper NameSpaces. You are addressing variables in the wrong Namespaces.

jpfr, 2018-11-18:
You can now add custom dataypes at runtime to the server config.
Enums are just int32 in the background.

closed 2018-11-18 by jpfr

juangburgos, 2019-03-20:
Hi guys,

I have been working on creating **custom enums** and came up with the following code that I leave below for reference, in case somebody else is looking to create custom enums.

**NOTE** that until #2584 is resolved, it is necessary to pass an explicit NodeId as a second argument to the `UA_Server_addDataTypeNode` method.

```c
#include <signal.h>
#include <stdio.h>
#include "open62541.h"

typedef struct
{
	UA_Int64         Value;
	UA_LocalizedText DisplayName;
	UA_LocalizedText Description;
} EnumValue;

static UA_StatusCode createEnumValue(const EnumValue * enumVal, UA_ExtensionObject * outExtObj)
{
	if (!outExtObj)
	{
		return UA_STATUSCODE_BADOUTOFMEMORY;
	}
	outExtObj->encoding               = UA_EXTENSIONOBJECT_ENCODED_BYTESTRING;
	outExtObj->content.encoded.typeId = UA_NODEID_NUMERIC(0, 8251);
	UA_ByteString_allocBuffer(&outExtObj->content.encoded.body, 65000);
	UA_Byte * p_posExtObj = outExtObj->content.encoded.body.data;
	const UA_Byte * p_endExtObj = &outExtObj->content.encoded.body.data[65000];
	// encode enum value
	UA_StatusCode st = UA_STATUSCODE_GOOD;
	st |= UA_encodeBinary(&enumVal->Value      , &UA_TYPES[UA_TYPES_INT64]        , &p_posExtObj, &p_endExtObj, NULL, NULL);
	st |= UA_encodeBinary(&enumVal->DisplayName, &UA_TYPES[UA_TYPES_LOCALIZEDTEXT], &p_posExtObj, &p_endExtObj, NULL, NULL);
	st |= UA_encodeBinary(&enumVal->Description, &UA_TYPES[UA_TYPES_LOCALIZEDTEXT], &p_posExtObj, &p_endExtObj, NULL, NULL);
	if (st != UA_STATUSCODE_GOOD)
	{
		return st;
	}
	size_t p_extobj_encOffset = (uintptr_t)(p_posExtObj - outExtObj->content.encoded.body.data);
	outExtObj->content.encoded.body.length = p_extobj_encOffset;
	UA_Byte * p_extobj_newBody = (UA_Byte *)UA_malloc(p_extobj_encOffset);
	if (!p_extobj_newBody)
	{
		return UA_STATUSCODE_BADOUTOFMEMORY;
	}
	memcpy(p_extobj_newBody, outExtObj->content.encoded.body.data, p_extobj_encOffset);
	UA_Byte * p_extobj_madatory_oldBody  = outExtObj->content.encoded.body.data;
	outExtObj->content.encoded.body.data = p_extobj_newBody;
	UA_free(p_extobj_madatory_oldBody);
	// success
	return UA_STATUSCODE_GOOD;
}

static UA_StatusCode addEnumValues(UA_Server * server, UA_NodeId * parent, const UA_UInt32 numEnumValues, const EnumValue * enumValues)
{
	// setup variable attrs
	UA_StatusCode retVal         = UA_STATUSCODE_GOOD;
	UA_VariableAttributes attr   = UA_VariableAttributes_default;
	attr.minimumSamplingInterval = 0.000000;
	attr.userAccessLevel         = 1;
	attr.accessLevel             = 1;
	attr.valueRank               = 1;
	attr.arrayDimensionsSize     = 1;
	attr.arrayDimensions         = (UA_UInt32 *)UA_Array_new(1, &UA_TYPES[UA_TYPES_UINT32]);
	if (!attr.arrayDimensions)
	{
		return UA_STATUSCODE_BADOUTOFMEMORY;
	}
	attr.arrayDimensions[0] = 0;
	attr.dataType = UA_NODEID_NUMERIC(0, UA_NS0ID_ENUMVALUETYPE);
[... 117 lines cut ...]
UA_Boolean running = true;
static void stopHandler(int sig) {
	UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
	running = false;
}

int main(void) {
	signal(SIGINT, stopHandler);
	signal(SIGTERM, stopHandler);

	UA_ServerConfig* config = UA_ServerConfig_new_default();

	UA_Server* server = UA_Server_new(config);

	addCustomEnumType    (server, "ThreatLevels");
	addCustomEnumVariable(server, "CurrentThreatLevel");

	UA_StatusCode retval = UA_Server_run(server, &running);
	UA_Server_delete(server);
	UA_ServerConfig_delete(config);
	return (int)retval;
}
```

[image]

davesliu, 2019-05-22:
@juangburgos 

Hi, Juangburgos 

I saw that your created enum type can be parsed by UaExpert.
Could we use similar way to create user defined structure？

juangburgos, 2019-05-23:
Hi @davesliu ,

For a user-defined structured I believe you are better off following the approach described in the [custom_datatype example](https://github.com/open62541/open62541/tree/master/examples/custom_datatype). 

But to my understanding, UAExpert might not be able to "read" your custom structure because both client and server need to be aware of the *encoding* used to serilize your custom type.

The fact that UAExpert can read my enum is because it is constructed by subtyping the known *Enumeration* type, and using the same encoding.

I am not an OPC UA expert, so take my advice with caution.