open62541/open62541 #2038: One event triggers two MonitoredItems

NIPE-SYSTEMS, 2018-09-03:
My client creates two MonitoredItems (different callbacks) targetting the same object (`i=42`). My server then triggers one of the monitored events (`i=1337`). open62541 executes both callbacks on the client-side indicating that both MonitoredItems received an event. I had expected that open62541 only executes the callback of the "correct" MonitoredItem.

1. Are my ids and select clauses correct? (I've not used `BaseEventType` because I want to distinguish the events.)
2. Am I missing something?
3. If everything works as expected, how is it possible to distinguish between two received events on the client-side?

Thanks in advance!

See below for code examples (omitted error handling, no errors occuring). This may be related to #2003. I'm using Visual Studio 2015 on Windows 10 with open62541 at af95b006494a23eff648bc40ff1cceaf3c15598a. /cc @aDogCalledSpot @Pro 

# Triggering on the server-side

```cpp
// assuming EventType with i=1337 and ObjectType with i=42
UA_NodeId nodeIdEvent;
UA_Server_createEvent(server, UA_NODEID_NUMERIC(ns, 1337), &nodeIdEvent);
UA_DateTime now = UA_DateTime_now();
UA_Server_writeObjectProperty_scalar(server, nodeIdEvent, UA_QUALIFIEDNAME_STATIC(0, "Time"), &now, &UA_TYPES[UA_TYPES_DATETIME]);
UA_String string = UA_STRING_STATIC("Hello World!");
UA_Server_writeObjectProperty_scalar(server, nodeIdEvent, UA_QUALIFIEDNAME_STATIC(ns, "Text"), &string, &UA_TYPES[UA_TYPES_STRING]);
UA_Server_triggerEvent(server, nodeIdEvent, UA_NODEID_NUMERIC(ns, 42), NULL, UA_TRUE);
```

# Create MonitoredItem that should trigger

```cpp
// assuming EventType with i=1337 and ObjectType with i=42
UA_MonitoredItemCreateRequest request = UA_MonitoredItemCreateRequest_default(UA_NODEID_NUMERIC(ns, 42));
request.itemToMonitor.attributeId = UA_ATTRIBUTEID_EVENTNOTIFIER;
UA_EventFilter filter;
UA_EventFilter_init(&filter);
filter.selectClauses = (UA_SimpleAttributeOperand *)UA_Array_new(1, &UA_TYPES[UA_TYPES_SIMPLEATTRIBUTEOPERAND]);

UA_QualifiedName Text = UA_QUALIFIEDNAME_ALLOC(ns, "Text");
UA_SimpleAttributeOperand_init(&filter.selectClauses[0]);
filter.selectClauses[0].typeDefinitionId = UA_NODEID_NUMERIC(ns, 1337); // NOTICE: triggered id
filter.selectClauses[0].browsePathSize = 1;
filter.selectClauses[0].browsePath = &Text;
filter.selectClauses[0].attributeId = UA_ATTRIBUTEID_VALUE;

// ... 3 more UA_SimpleAttributeOperand

filter.selectClausesSize = 1;
request.requestedParameters.filter.encoding = UA_EXTENSIONOBJECT_DECODED;
request.requestedParameters.filter.content.decoded.data = &filter;
request.requestedParameters.filter.content.decoded.type = &UA_TYPES[UA_TYPES_EVENTFILTER];
UA_Client_MonitoredItems_createEvent(client, subId, UA_TIMESTAMPSTORETURN_SOURCE, request, NULL, &callbackShouldTrigger, NULL);
```

# Create MonitoredItem that should NOT trigger

```cpp
// assuming other EventType with i=2337 and same ObjectType with i=42
UA_MonitoredItemCreateRequest request = UA_MonitoredItemCreateRequest_default(UA_NODEID_NUMERIC(ns, 42));
request.itemToMonitor.attributeId = UA_ATTRIBUTEID_EVENTNOTIFIER;
UA_EventFilter filter;
UA_EventFilter_init(&filter);
filter.selectClauses = (UA_SimpleAttributeOperand *)UA_Array_new(4, &UA_TYPES[UA_TYPES_SIMPLEATTRIBUTEOPERAND]);

UA_QualifiedName Status1 = UA_QUALIFIEDNAME_ALLOC(ns, "Status1");
UA_SimpleAttributeOperand_init(&filter.selectClauses[0]);
filter.selectClauses[0].typeDefinitionId = UA_NODEID_NUMERIC(ns, 2337); // NOTICE: different id (not triggered)
filter.selectClauses[0].browsePathSize = 1;
filter.selectClauses[0].browsePath = &Status1;
filter.selectClauses[0].attributeId = UA_ATTRIBUTEID_VALUE;

// ... 3 more UA_SimpleAttributeOperand

filter.selectClausesSize = 4;
request.requestedParameters.filter.encoding = UA_EXTENSIONOBJECT_DECODED;
request.requestedParameters.filter.content.decoded.data = &filter;
request.requestedParameters.filter.content.decoded.type = &UA_TYPES[UA_TYPES_EVENTFILTER];
UA_Client_MonitoredItems_createEvent(client, subId, UA_TIMESTAMPSTORETURN_SOURCE, request, NULL, &callbackShouldNotTrigger, NULL);
```

aDogCalledSpot, 2018-09-03:
This appears to be related to https://github.com/open62541/open62541/issues/2027 and should be fixed soon.

NIPE-SYSTEMS, 2018-09-17:
@aDogCalledSpot What is the status of #2027? The issue is still open but there isn't any activity nor comments within two weeks about the status.

PR #2062 (2018-09-22): Correct Event validity check
Make sure the EventType is not a Subtype of CondtionType (Part 9 not supported yet).
1) The check should ignore the "ConditionId" Clause set by UaExpert and ignore all Events which are Conditions. (However, if a Condition Event was created and UaExpert is used, then all the Fields of that Event except the "ConditionId" will be presenet in the Event View) -> The User should not use Events which are Conditions in that case!
2) The second check is to make sure the Event has a TypeDefinition of an BaseEventType SubType.

Fixes #2027 and fixes #2038 

Cherry-picked from [... cut]
closed 2018-09-22 by jpfr