open62541/open62541 #2043: Build Error in MinGW32

cmbahadir, 2018-09-04:
Description
===========

Hi,

I am trying to build on MinGW32 but it throws "-Werror=sign-conversion" for UA_fd_set().

Background Information / Reproduction Steps
===========================================

CMake options;

[image]

Used MinGW headers are (comes with installation by default);

- _socket_types.h :
[_socket_types.txt](https://github.com/open62541/open62541/files/2347862/_socket_types.txt)

- winsock2.h 
[winsock2.txt](https://github.com/open62541/open62541/files/2347863/winsock2.txt)

- _fd_types.g:
[_fd_types.txt](https://github.com/open62541/open62541/files/2347864/_fd_types.txt)

**Result:**

ua_network_tcp.c compile fails as;
/open62541/arch/ua_network_tcp.c:90:9: error: conversion to ‘long unsigned int’ from ‘long int’ may change the sign of the result [-Werror=sign-conversion]
UA_fd_set(connection->sockfd, &fdset);

and 

/d/Task_Force/open62541/arch/ua_network_tcp.c: At top level:
cc1: error: unrecognized command line option ‘-Wno-static-in-inline’ [-Werror]

make --trace log
[mingw32_make_trace.txt](https://github.com/open62541/open62541/files/2347826/mingw32_make_trace.txt)

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (latest): 0.3-rc2 - 93c4d6caf2ff660042f7a46e1fc46cd3712a95bf
 - [x] Operating system: Windows - MinGW32 - gcc version 6.3.0
 - [x] Logs (with UA_LOGLEVEL set as low as necessary) attached: see mingw32_make_trace.txt

Stasik0, 2018-09-14:
https://github.com/open62541/open62541/blob/master/appveyor.yml

Every build is compiled using mingw. Did you try mingw64?

cmbahadir, 2018-09-26:
It seems build fails on some of the mingw versions. As also mentioned on this issue 

https://github.com/open62541/open62541/issues/1734#issuecomment-380711629

So ı close it. Thanks.

closed 2018-09-26 by cmbahadir