open62541/open62541 #2044: Development package

kjeldflarup, 2018-09-04:
I would like to add open62541 to my OE-lite (an embedded toolchain)  
For this I would like to create a development package and add it to the SDK I generate. This however does not seem to be supported in open62541

If I take the full install with shared libraries
```
cmake -DBUILD_SHARED_LIBS=true .
make
make DESTDIR=/tmp/open62541 install
```
Then `open62541.h` is missing in the install. But the `nodeset_compiler` and `cmake` stuff is there.  
Is it enough to add open62541.h or are there more things?

Also, when installing on an embedded target I do definitely not want anything else than the .so files. So I need to be able to define an open62541 and an open62541-devel package.

kjeldflarup, 2018-09-04:
To me it looks like the current install is what should go into the devel package, except for the .so files.  
 #1415 should clearly be fixed.  
But #1480 may contain some caveats.

We are just starting out implementing OPC UA support in our target. The tests for now may be done with the amalgamated approach, but if we go live with open62541 we will have to fix this.

traversaro, 2018-09-04:
I may say something obvious, but I am in similar situation to you, I use `open62541` as part of CMake superbuild, and I think building the project as in:
~~~
cmake -DBUILD_SHARED_LIBS=true -DUA_ENABLE_AMALGAMATION:BOOL=ON .
make
make DESTDIR=/tmp/open62541 install
~~~ 
should fix (for now) your problem with the missing `open62541.h` . 

Clearly this does not substitute the cleanup discussed in https://github.com/open62541/open62541/issues/1415 and https://github.com/open62541/open62541/issues/1480 .

jpfr, 2018-09-06:
Dear all,

the open62541.h is an artefact of the "amalgamated build" that creates a single open62541.c and open62541.h. If you use cmake to build, you can directly use for example `/include/ua_server.h`. No need to use open62541.h.

In the long-term, I propose to use a different scheme for the public headers.
If we rename `/include/ua_server.h` to `/include/open62541/server.h`, then it is a lot simpler to install the headers on a Linux box. And the headers are used the same, regardless of whether we refer to the source dir or the installed headers by some distribution package.

traversaro, 2018-09-06:
Great, I was not aware of this, especially because all the tutorials only refer `#include "open62541.h"`. 

Do you think that, to reduce the complexity in code switching from using the "amalgamated" and the regular build, it would make sense to have the same header name for the "public" header in the regular build and header of the amalgamated build?

traversaro, 2018-09-06:
For example, this can achieved in regular build by generating a `open62541.h` header that just includes the `ua_server.h` header.

traversaro, 2018-11-14:
Note that the "trick" of enabling `` to install a `` is used also in [`basyskom`](https://www.basyskom.com/) docs on how to install open62541 with Qt OPC UA : https://blog.basyskom.com/building-qt-opc-ua-with-open62541/

traversaro, 2018-12-05:
Thanks @Pro and @aDogCalledSpot for solving this! 

I have just a small question after https://github.com/open62541/open62541/pull/2265 : if I understood correctly, now the amalgamated header `open62541.h` is installed also if  `UA_ENABLE_AMALGAMATION` is OFF (that is the default situation), see https://github.com/open62541/open62541/blob/102b7dc69395a9f6e93cd35884b1295f30c6d2d4/CMakeLists.txt#L1051 . 
However, now also the discrete headers such as `open62541/ua_server.h` are installed. As the amalgamated header does not contain the discrete header include guards, this means that if a compilation unit contains both `#include <open62541.h>` and   `#include <open62541/ua_server.h>` (note that this can happen also due to transitive includes) you will get errors due to multiple function definitions.

traversaro, 2018-12-05:
> So one should either include the open62541.h OR all the necessary files in <open62541/ua_server.h>

This make sense: however if you are integrating several libraries that use `open62541` in their public headers, if both way of including the headers are supported, it is possible that one library uses `<open62541.h>`, while another uses `<open62541/ua_server.h>`, making impossible to use them in the same compilation unit without modifying their code. I think we can avoid this problem if the `open62541.h` installed in the non-amalgamated case included the necessary single headers, such `#include <open62541/ua_server.h>` or any other necessary header, instead of using the amalgamated `open62541.h`.

Pro, 2018-12-05:
The recommended include is `open62541.h` and not the single files. If you mix both you will never be able to compile both compilation units together because you will be missing one header or the other.

Including the other headers in the main open62541.h will also not work since we want to have a single file distribution where you have two files, a .h and one .c containing all the code.

The only possible solution I see is to not remove the ifdef guards. But maybe it's better to keep it as it is, since including/installing the single header is not recommended.

Pro, 2018-12-05:
Yes, you are right, i'm also not so happy about that...

@jpfr should we install separate header files, or only the amalgamated open62541.h, independent of the UA_AMALGAMATION setting?

kjeldflarup, 2018-12-05:
Say You wanted to make packages for Debian. How would that be most logical?

I'm actually backing out of installing open62541 into my SDK, as there are so many different options that the library can be compiled with. Thus my approach is to compile my application with open62541.

Pro, 2018-12-05:
The recommended packaging is to enable all non-experimental features of the release and also the full ns0, with amalgamation enabled.

Pro, 2018-12-05:
But packaging is still not optimal!
We are happy for any recommendations

Pro, 2019-01-03:
@kjeldflarup and @traversaro:
In the last weeks we fixed many cmake install issues and now we have a more or less clean installation.

Also check out the docs:
https://open62541.org/doc/current/installing.html

Note that we recommend to install the NON-amalgamated version, which means that there will not be any "open62541.h" file, but the separate public headers.
You may also add the `UA_ENABLE_AMALGAMATION=ON` option to cmake for installation, which will only install the amalgamated open62541.h header.

Does that work for you? Please reopen if questions remain.

Note that there is currently an open issue where we still need to come to a conclusion what the best way is to organize the public headers: #2309

closed 2019-01-03 by Pro

traversaro, 2019-01-03:
Thanks a lot for the efforts @Pro . I will test the new improvements as soon as we update the version of open62541 that we use inside the project I am working on, and I will report any eventual problem.

PR #2262 (2018-12-02): Install without amalgamation
Same as #2260 but for master

Fixes #1415 
Fixes #1480 
Fixes #2044