open62541/open62541 #2051: Could not remove monitored item 1 with error code BadConnectionClosed

vdeters, 2018-09-11:
Description
===========
Hello, the error message in the subject line appears when my client calls UA_Client_Subscriptions_removeMonitoredItem(). I have a Wireshark capture and the only thing it shows around that time is a FIN packet from client to server. It does show a prior CreateMonitoredItemsResponse packet showing a good result for MonitoredItemId 1. 

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.2-rc2
 - [x] Other OPC UA SDKs used (client or server): server: Yokogawa Fast/Tools server running (version UA2.0) on Windows Server 2012 R2
 - [x] Operating system: client: linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Wireshark network dump attached: set this filter: ip.addr==10.70.232.207, see disconnect at line 403
 - [ ] Self-contained code example attached
 - [x] Critical issue

[pcap.tar.gz](https://github.com/open62541/open62541/files/2372235/pcap.tar.gz)

vdeters, 2018-09-13:
> Hello @vdeters <https://github.com/vdeters> ,
>
> I had a look the the capture. Thank you for providing the detailed
> information.
>
>    - Does the same behaviour occur with the current 0.2 branch or was the
>    problem already fixed?
>    - Do you see any client-side log output? You may have to enable
>    trace-level logs...
>
> [image: esp-pcap]
> <https://user-images.githubusercontent.com/1726411/45490796-5ed06e80-b768-11e8-81b9-62af337bfd11.png>

We don't know if it works with 0.2 because we can't reproduce it here. For
that reason I suspect it may be related to something that specific server
is doing. Can you verify that the client/server interchange in the capture
looks standard and correct?

How do I enable trace-level logs? I assume I'll have to recompile our
client app, update the customer, and have them retry.

Vince

vdeters, 2018-09-25:
We rebuilt our client against open62541 version 0.2 and had the customer retry the same scenario, and results were the same. New Wireshark capture attached (25sep.gz). As before, apply filter ip.addr==10.70.232.207 and see the FIN sent from the client at the end. With default logging, the client emits this:

[09/25/2018 11:23:09.235] info/client   Could not remove monitoreditem 1 with error code BadConnectionClosed

The error was triggered by a call to UA_Client_Subscriptions_removeMonitoredItem(), and calling UA_StatusCode_description() on the return value returned "The network connection has been closed."

Please give this some prompt attention. Thanks.

[25sep.gz](https://github.com/open62541/open62541/files/2415882/25sep.gz)

jpfr, 2018-09-29:
> Somebody please respond, this is marked as critical. Thanks.

Please do not mistake this community to provide "free labour".
We try to be helpful. But resources are limited.
This is an open source. Nothing prevents you from fixing the bug yourself and sending us a patch.

If you need application-specific assistance, there are companies doing exactly that for open62541.

On a different note, I added a unit test to reproduce the behaviour: #2088
In the test, RemoveMonitoredItems works fine for the 0.2 branch.
Did you try to upgrade from the release candidate to the 0.2 release or (even better) the current 0.2 branch?

jpfr, 2018-09-29:
> Do you see any reason for the disconnection in the most recent wireshark capture? 

The 25sep.gz file cannot be opened with Wireshark. Which type of file is that?

> I understand I can fix it myself, but I can’t reproduce it against our test server (as noted above). 

We are in the same position. The original wireshark dump just shows that the client closes the socket without sending a CLO message.
That could be anything. The client crashing, the operating system closing the process, etc.

Your best way forward is to add instrumentation. Set UA_LOGLEVEL to 100, attach a debugger, etc.

vdeters, 2018-09-30:
25sep.gz is a gzipped file. You can uncompress it on linux systems with tar -zxvf.

jpfr, 2018-09-30:
> 25sep.gz is a gzipped file. You can uncompress it on linux systems with tar -zxvf.

Ah. This is a tarball and not just gzipped.
I had to rename the file to .tar.gz to open it.

The capture is essemtially the same as the first. So nothing new.

How do you know the error is caused by a call to `UA_Client_Subscriptions_removeMonitoredItem` if you could not reproduce?

vdeters, 2018-10-01:
Because my client logged an error specific to receiving a failure return code from UA_Client_Subscriptions_removeMonitoredItem().

jpfr, 2018-10-01:
The client says
```
[09/25/2018 11:23:09.235] info/client Could not remove monitoreditem 1 with error code BadConnectionClosed
```

But from the wireshark dump we see that the client is the one closing the connection.
Is there any more interesting log information above the line you pasted?

vdeters, 2018-10-01:
Here is the complete console from my client around the time of the error:

[09/25/2018 11:23:09.235] info/client   Could not remove monitoreditem 1
with error code BadConnectionClosed
2018-09-25T16:53:09,235; ERROR; 00000018; DF.ESP;
(dfESPopcuaConnector.cpp:1171); [Connectors0045]
dfESPopcuaConnector::pubThread():
UA_Client_Subscriptions_removeMonitoredItem() failed, error message = The
network connection has been closed.
[09/25/2018 11:23:09.236] info/client   Could not remove monitoreditem 1
with error code BadConnectionClosed

The 2nd line is logged by my code immediately after receiving the bad
return code from UA_Client_Subscriptions_removeMonitoredItem(), and it
includes the string returned by calling UA_StatusCode_description() on the
return value. There is no other console output from the open62541 library.

So from this I presume that open 62541 logs "Could not remove..." from
UA_Client_Subscriptions_removeMonitoredItem(), which then returns the bad
return code. It's possible that the closed connection we see in the
Wireshark happened sometime earlier for an unrelated reason - could it be
something as simple as a missed heartbeat?

vdeters, 2018-10-03:
I just discovered this case:
https://github.com/open62541/open62541/issues/1485

I interpret the issue as follows (for 0.2):
UA_ClientConfig_default defines some default timeout of 5 seconds for publish request/responses. If I call UA_Client_Subscriptions_manuallySendPublishRequest() and there is no value change within this timeout period the client is disconnected. Is this correct? If so, is there any workaround using 0.2?

vdeters, 2018-10-11:
Further update: we rebuilt our client against open62541 version 0.3rc2, and it revolved this issue.

teccheck, 2025-04-15:
As of @vdeters last comment, I looks like this issue is resolved

closed 2025-04-15 by teccheck