open62541/open62541 #2052: Core dump on closing connection

kjeldflarup, 2018-09-12:
Description
===========
I have written some servers and clients with open62541 and found that it often coredumps.

Background Information / Reproduction Steps
===========================================
I boiled it down to this example on my Ubuntu 16.04
```
git checkout master
git pull
cmake -DUA_ENABLE_MULTITHREADING=true -DUA_ENABLE_AMALGAMATION=true .
make
gcc -g -pthread -std=c99 -DUA_ARCHITECTURE_POSIX simpleclient.c open62541.c -o simpleclient
gcc -g -pthread -std=c99 -DUA_ARCHITECTURE_POSIX simpleserver.c open62541.c -o simpleserver
```
simpleserver.c 
```
#include "open62541.h"
UA_Boolean running = true;
int main(void) {
    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_Server *server = UA_Server_new(config);
    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return (int)retval;
}

```
simpleclient.c
```
#include "open62541.h"
int main(void) {
    UA_Client *client = UA_Client_new(UA_ClientConfig_default);
    UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:4840");
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        return (int)retval;
    }
    UA_Client_delete(client); /* Disconnects the client internally */
    return UA_STATUSCODE_GOOD;
}
```
Running simpleclient multiple times will provoke the core dump.

```
$ gdb ./simpleserver 
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./simpleserver...done.
(gdb) r
Starting program: /userdata/opc/server/coredump/simpleserver 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[2018-09-12 10:31:39.273 (UTC+0200)] info/network	TCP network layer listening on opc.tcp://kjeld-Latitude-5580:4840/
[2018-09-12 10:31:39.273 (UTC+0200)] info/server	Spinning up 1 worker thread(s)
[New Thread 0x7ffff77ef700 (LWP 19765)]
[2018-09-12 10:31:45.172 (UTC+0200)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[... 71 lines cut ...]
The error seems to be in UA_SecureChannel_deleteMembersCleanup 
```
19157	    /* Detach from the connection and close the connection */
19158	    if(channel->connection) {
19159	        if(channel->connection->state != UA_CONNECTION_CLOSED)
19160	            channel->connection->close(channel->connection);
19161	        UA_Connection_detachSecureChannel(channel->connection);
19162	    }
```
channel->connection is 0x0 when the error happens, but it certainly wasn't when entering the if statement in line 19158

Could it be related to multithreading?  
I need that to keep the server running, as my application has other tasks than doing OPC UA

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2018-09-12:
Thank you for the bug report.

Since you are using a very experimental feature, i.e. `UA_ENABLE_MULTITHREADING` segfaults and core dumps are expected behaviour.

We did not test multithreading support for a long time since we are currently focusing on other more important features. There are many places in the code where multithreading supports needs to be added.

I suggest that you disable multithreading and make sure that you are using your own mutexes in user code if you need multithreading support.

We can keep this issue open, maybe someone is willing to contribute a PR

kjeldflarup, 2018-09-12:
That leaves me with a question, how do I run the server in a thread, and set the data from another thread?  
There is a setting UA_ENABLE_IMMUTABLE_NODES, is that what makes it possible. And does that have a cost?

Infinity95, 2018-09-13:
I don't know anything about the immutable nodes, but you could maybe take the following approach?:

Have one thread that runs the open62541 server and which accepts jobs/data in a queue from other threads.
Have other threads put jobs/data into the open62541 server thread queue.
If the server thread needs to produce output, put it into some kind of structure and put it into an output queue that can be emptied from other threads. If you need an association between results and threads, you could have one output queue for each thread, and put the result in the appropriate one.
The only inter-thread communication would occur through the queues which reduces potentials for errors, and there should be a lot of thread safe queue implementations available, i think.

I don't know how much this will help you. Just wanted to share my thoughts.

jpfr, 2018-09-13:
At this time, you should use a mutex to protect the server from parallel access.
We have the architecture basics for multithreading. But this is nowhere production ready.

jpfr, 2018-09-13:
@Infinity95 
It's nice to have a consensus on how multithreading should work.
You describe the current implementation. 😄 
https://github.com/open62541/open62541/blob/master/src/server/ua_server_internal.h#L163

But there are some touch-points where threads interact.
Information model concurrency is sound.
Parallel access to Sessions, SecureChannels and connections is not.

Infinity95, 2018-09-13:
Yeah, what i meant was to abstract even further and treat the opc ua server as a single threaded application, and just post jobs to this single threaded application.

jpfr, 2018-09-15:
Ok. So this is a wontfix.
Please use a mutex to prevent parallel access to the client/ server.

Multithreading will be finalized eventually. But it is explicitly marked as experimental for now.

closed 2018-09-15 by jpfr