open62541/open62541 #2057: CTT issue on 0.3 branch: OperationLimits_MaxNodesPerMethodCall shall not be present if methods are not supported

jonasgreen88, 2018-09-14:
Description
===========
We have run into a CTT issue that we believe exists in the stack when method calls are disabled. Even if method calls are disabled in the stack the MaxNodesPerMethodCall in the OperationLimits of the Server object is present in the address space. When this node is present in the address space of the server the CTT expects the method GetMonitoredItems to be implemented, which it obviously isn't when method calls are disabled.  If method calls are disabled in the stack, this node must be removed.
The issue have been reported and discussed on the OPC Foundation issue tracker: https://opcfoundation-onlineapplications.org/mantis/view.php?id=4387.

I looked into the nodeset compiler and found out that all nodes of type method or methodtype are filtered away if method calls are disabled. I'm not really sure how the MaxNodesPerMethodCall shall be filtered away as it is a normal variable node. Do you guys have any idea?

Background Information / Reproduction Steps
===========================================
Run CTT test case "Base Information - Base Info ServerCapabilities - 014" on a server with method calls disabled.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0c7ccfd (latest 0.3 branch)
 - [ ] Other OPC UA SDKs used (client or server): CTT Compliance Test Tool 1.03.340.380
 - [ ] Operating system: Windows 7
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jonasgreen88, 2018-09-14:
This problem is probably also valid for the MaxNodesPerNodeManagement (UA_ENABLE_NODEMANAGEMENT) and MaxMonitoredItemsPerCall (UA_ENABLE_SUBSCRIPTIONS) nodes of the OperatingLimits node as well.

Pro, 2018-09-14:
If I read through the OPCF issue tracker, I parse the following:
- `MaxNodesPerMethodCall` can still be there, even if method calls are disabled. And if it is there the value has to be > 0
- If method calls are not supported, then the server should return "UA_BAD_NOTIMPLEMENTED" for any `Service_call`

I quickly checked, we are returning BAD_SERVICEUNSUPPORTED for the Service_call. Is that correct? Or should it be NOTIMPLEMENTED? Does that fix the CTT?

Otherwise we may need to add an additional list in the nodeset compiler which has a list of nodeids which should be wrapped in an ifdef UA_ENABLE_METHODCALLS.

jonasgreen88, 2018-09-14:
What I can see when looking into the test script it looks like the CTT expects the service to work if the MaxNodesPerMethodCall node is present. But I will confirm this on monday when I can run the test.

jonasgreen88, 2018-09-21:
> If I read through the OPCF issue tracker, I parse the following:
>
> * `MaxNodesPerMethodCall` can still be there, even if method calls are disabled. And if it is there the value has to be > 0
> * If [...]

@Pro We got further feedback on our issue on the CTT (https://opcfoundation-onlineapplications.org/mantis/view.php?id=4387). It should be ok to have the MaxNodesPerMethodCall even if the method GetMonitoredItems is not supported, but the value must be != 0. They also say that the error code should be NOTIMPLEMENTED. So I guess we should change the error code the stack returns but we don't have to remove the node.

jonasgreen88, 2018-09-27:
@Pro Is this the error code that you think should be changed to NOTIMPLEMENTED?
https://github.com/open62541/open62541/blob/c9c2088124d6b3ff4e3068921812f1abdac40536/src/server/ua_server_binary.c#L434

I

closed 2023-09-18 by jonasgreen88