open62541/open62541 #2058: CTT issue: Value attribute of SoftwareCertificates node not default empty

jonasgreen88, 2018-09-14:
Description
===========
The CTT complains about that the value of SoftwareCertificates node of the Server object. Our device has no software certificates so this node should be empty. The following error code is given by CTT when running test Base Info Server Capabilities test case 000: "Error. Expected a value less that <1> but got <1>. Expected SoftwareCertificates to be empty.".

The nodeset compiler seems to add a 1 element to the value attribute of the SoftwareCertificates node when it generates the namespace0.c file, even if the array length given in the first element of the arrayDimensions attribute is 0.

Below is the complete SoftwareCertificates function of ua_namespace0.c:
```
/* SoftwareCertificates - ns=0;i=3704 */

static UA_StatusCode function_ua_namespace0_94_begin(UA_Server *server, UA_UInt16* ns) {

UA_StatusCode retVal = UA_STATUSCODE_GOOD;
UA_VariableAttributes attr = UA_VariableAttributes_default;
attr.minimumSamplingInterval = 0.000000;
attr.userAccessLevel = 1;
attr.accessLevel = 1;
attr.valueRank = 1;
attr.arrayDimensionsSize = 1;
attr.arrayDimensions = (UA_UInt32 *)UA_Array_new(1, &UA_TYPES[UA_TYPES_UINT32]);
attr.arrayDimensions[0] = 0;
attr.dataType = UA_NODEID_NUMERIC(ns[0], 344);
UA_STACKARRAY(UA_SignedSoftwareCertificate, variablenode_ns_0_i_3704_variant_DataContents,1);
UA_init(&variablenode_ns_0_i_3704_variant_DataContents[0], &UA_TYPES[UA_TYPES_SIGNEDSOFTWARECERTIFICATE]);
UA_Variant_setArray(&attr.value, variablenode_ns_0_i_3704_variant_DataContents, (UA_Int32) 1, &UA_TYPES[UA_TYPES_SIGNEDSOFTWARECERTIFICATE]);
attr.displayName = UA_LOCALIZEDTEXT("", "SoftwareCertificates");
attr.description = UA_LOCALIZEDTEXT("", "The software certificates owned by the server.");
attr.writeMask = 0;
attr.userWriteMask = 0;
retVal |= UA_Server_addNode_begin(server, UA_NODECLASS_VARIABLE,
UA_NODEID_NUMERIC(ns[0], 3704),
UA_NODEID_NUMERIC(ns[0], 0),
UA_NODEID_NUMERIC(ns[0], 0),
UA_QUALIFIEDNAME(ns[0], "SoftwareCertificates"),
UA_NODEID_NUMERIC(ns[0], 68),
(const UA_NodeAttributes*)&attr, &UA_TYPES[UA_TYPES_VARIABLEATTRIBUTES],NULL, NULL);
UA_Array_delete(attr.arrayDimensions, 1, &UA_TYPES[UA_TYPES_UINT32]);
retVal |= UA_Server_addReference(server, UA_NODEID_NUMERIC(ns[0], 3704), UA_NODEID_NUMERIC(ns[0], 46), UA_EXPANDEDNODEID_NUMERIC(ns[0], 2268), false);
return retVal;
}
```
The issue is on this line: 
```
UA_Variant_setArray(&attr.value, variablenode_ns_0_i_3704_variant_DataContents, (UA_Int32) 1, &UA_TYPES[UA_TYPES_SIGNEDSOFTWARECERTIFICATE]);
```

The CTT pass the test if that line is changed to: 
```
UA_Variant_setArray(&attr.value, NULL, (UA_Int32) 0, &UA_TYPES[UA_TYPES_SIGNEDSOFTWARECERTIFICATE]);
```

If looking at the SoftwareCertificates node in Opc.Ua.Nodeset2.Minimal.xml there is no value defined:
https://github.com/open62541/open62541/blob/0c7ccfde82886d4e7c9bce4d7af5c2078f9ae2ff/tools/schema/Opc.Ua.NodeSet2.Minimal.xml#L1259

Looking into the nodeset compiler and comparing its behavior with the generated code, it seems like the script takes the wrong path at this point: https://github.com/open62541/open62541/blob/c5d730af2983ed1dbfad930819776c1ef72fee65/tools/nodeset_compiler/backend_open62541_nodes.py#L114

As the node has no value defined in the xml file it should enter the else clause I think, and in that case it should generate the code that made CTT to pass. But instead the nodeset compiler seems to validate the if statement to true and execute the if clause rather than the else clause.

Background Information / Reproduction Steps
===========================================
Run Base Info Server Capabilities test case 000 on a server device that doesn't have any certificates installed.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0c7ccfd (latest 0.3 branch)
 - [ ] Other OPC UA SDKs used (client or server):  CTT 1.03.340.380
 - [ ] Operating system: Windows 7
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2018-09-14:
Thanks for digging into it!
I am using CLion (from JetBrains, or PyCharm is also nice) to debug into the nodeset compiler to find such issues. You can also use any other python debugger. The command line args you need for the nodeset compiler can be found into the generated CMake files in your build directory. Just search for `nodeset_compiler.py`

I would have time next week to look into it and try to find the issue.

Pro, 2018-09-14:
I just made a commit in a hotfix branch `hotfix/empty_value_nodeset`:
https://github.com/open62541/open62541/commit/2cbad9e6774af94581c919c48f32ce5962cb38e2

Can you try with that change, if it fixes the issue?

The code is executing the `generateValueCodeDummy` for empty values, but it is setting there an empty value for the variant (array). IMHO the value should also be NULL here. I'm not yet 100% sure why we were setting a value here in the first place... . 

@jpfr  If this doesn't open any new issues, then I would just create a PR and merge it into 0.3 and master

jonasgreen88, 2018-09-17:
Hi,

Thanks for quick response on this issue!

I have now tried this solution and the nodeset compiler generates null values to the nodes as expected. However, I get issues when initializing the these nodes. When adding a node the function AddNode_typeCheckAddRefs() is called. That function then calls typeCheckVariableNode(). In that function, this if statement changes arrayDimensions to 1:
https://github.com/open62541/open62541/blob/0c7ccfde82886d4e7c9bce4d7af5c2078f9ae2ff/src/server/ua_services_nodemanagement.c#L177
Which makes the next if statment to return BADTYPEMISMATCH:
https://github.com/open62541/open62541/blob/0c7ccfde82886d4e7c9bce4d7af5c2078f9ae2ff/src/server/ua_services_nodemanagement.c#L183

I can see that this implementation has slightly changed on the master branch. Maybe those changes solve this issue. But if this solution shall be merged to 0.3 (which would be nice) some more changes must be merged I guess.

jonasgreen88, 2018-10-08:
@Pro @jpfr 

Hi, I did some more investigation with this hotfix on the 0.3 branch and got it working after cherry-picking two other commits from the master branch:

- 83a83410cb93d5ad861fb4da906651442aaff298
- 5fa9e9b2f76db7d0f94b4293ab52989a6651e988

So this fix probably works as is on the master branch but needs two additional fixes from the master branch to work on the 0.3 branch.

jpfr, 2018-10-09:
@staldert provided the cherry-pick PR. Thanks!

Ok to close?

jonasgreen88, 2018-10-10:
> @StalderT provided the cherry-pick PR. Thanks!
> 
> Ok to close?

Thanks for the help merging to the 0.3 branch. I have downloaded and tested the fix again and it works fine!

commit cce2d622 (2018-10-09): Do not set the value of a variant if the node does not have a value in the nodeset. Fixes #2058
closed 2018-10-10 by jonasgreen88