open62541/open62541 #2061: [Feature request] add functions to deal with custom data types etc.

undroidwish, 2018-09-15:
Please review http://www.androwish.org/index.html/dir?ci=14776af02a89dace&name=jni/topcua and especiallly http://www.androwish.org/index.html/artifact/35ccc53348a99429 for what was required to make things in open62541 somewhat more handy for an interface to a dynamic language in order to simplify C interfacing regarding basic data type mapping and custom data type handling.

jpfr, 2018-09-16:
Improving type handling is a good idea.
In principle, the custom datatypes array can be accessed via `UA_Client_getConfig` already.

In addition, we could make the "custom" datatypes a linked list (of datatype arrays).
So we can add custom datatypes during initialization.
A `realloc` should be avoided because there could be pointers into the custom datatypes array already.

undroidwish, 2018-09-17:
Thank you. The point with the custom data type arrays is to be able to deal with them during runtime, i.e. not when the (client|server) constructor is called, but always. A problem is of course the lifetime of variables which use those definitions. But when this feature is used to only add definitions later no inconsistencies should occur. And a language binding might even enforce addition only.

jpfr, 2018-11-18:
This was merged in #2085.
So you can now add datatypes at runtime with a public API.
Of course, the type descriptions should not be freed as long as there are values the still reference them.

closed 2018-11-18 by jpfr