open62541/open62541 #2063: SourceTimestamp of initialized data

kleskjr, 2018-09-17:
Description
===========
When reading a node from a open62541 server, normally it has `SourceTimestamp` value that reflects the time of the last write. However if I read from a node that has never been writen into, I get the current timestamp. I was wandering if this is an issue that others have stumbled at. Or is it some obsecure point in the opc-ua specifications? Is there a way to initialize the `SourceTimestamp` of all the nodes during serve execution?

Background Information / Reproduction Steps
===========================================

Server is generated through a xml file using the python-nodeset compiler. The node values are initialized in the xml. If some client writes to node A, then node A has a timestamp of the last write. If there is another node B, and no client ever wrote into it, then its timestamp corresponds to the time of the reading. In my opinion such implementation doesn't make much sence, but maybe there are some arguments for such behaviour that I can't forsee.

Checklist
=========
Please provide the following information:

 - [ 0.3-rc2 ] open62541 Version (release number or git tag):
 - [ server, nodeset_compiler] Other OPC UA SDKs used (client or server): 
 - [ Ubuntu ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ No ] Critical issue

Pro, 2018-09-18:
When reading, the timestamp is set to current here if the node does not yet have a source timestamp:
https://github.com/open62541/open62541/blob/0c3cbfba9f2039375208af2d86dd6358fc30f8b4/src/server/ua_services_attribute.c#L372

When writing it is set here:
https://github.com/open62541/open62541/blob/0c3cbfba9f2039375208af2d86dd6358fc30f8b4/src/server/ua_services_attribute.c#L1097

The Spec Part 4 says:
[image]

As far as I can see it, we should not use the current time, but probably the server start time if the variable does not yet have a source timestamp, i.e. it was initialized at server startup? @jpfr

kleskjr, 2018-09-19:
Thanks for the reply! Yep, right. If a node is initialized then it makes sense to return the initialization (server startup) timestamp. What would happen with nodes that are not initialized?

jpfr, 2018-10-22:
If the variable does not have a SourceTimestamp, then it will be disabled in the DataValue.
The DataValue has a flag to indicate for each timestamp if it is present.

closed 2018-10-22 by jpfr