open62541/open62541 #2068: CI Appveyor build fails

hthiery, 2018-09-18:
Description
===========

the CI build for PR fails

Background Information / Reproduction Steps
===========================================

Environment: CC_NAME=MinGW Makefiles, CC_SHORTNAME=mingw, MAKE=mingw32-make -j1 2>&1, FORCE_CXX=OFF, OUT_DIR_LIB=bin, OUT_DIR_EXAMPLES=bin\examples

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

hthiery, 2018-09-18:
maybe it is dependent on version to be installed?

PASS: `Packages (1) mingw-w64-x86_64-mbedtls-2.9.0-1`
FAIL:    `Packages (1) mingw-w64-x86_64-mbedtls-2.12.0-1`

Pro, 2018-09-18:
It also passed with 2.12 here:
https://ci.appveyor.com/project/open62541/open62541/build/7210/job/6atpoww1yuqkeqio?fullLog=true

I fixed the issue here, will create a PR for master and 0.3:
https://github.com/Pro/open62541/commit/58f9a7be56ab689e6889db1fa6460040c0d420ca

commit a2e52fb1 (2018-09-18): Fix pacman cache dir missing. Fixes #2068
commit cfd80512 (2018-09-18): Fix pacman cache dir missing. Fixes #2068
commit 2857d31f (2018-09-18): Fix pacman cache dir missing. Fixes #2068