open62541/open62541 #2074: CERN uses open62541

fabian727, 2018-09-20:
I do not know what they are using of this stack (or if they still use it), but 
CERN has internal servers to bridge canopen over OPC UA and state they use open62541
[github](https://github.com/quasar-team/quasar)
[docu](https://readthedocs.web.cern.ch/display/ICKB/OPC-UA)
[artikel i found about it](https://opcconnect.opcfoundation.org/2015/06/opc-news/)

just wanted you to know about it. Maybe you can reference their repo in your wiki

jpfr, 2018-09-21:
Thanks for pointing this out.

Actually, OPC UA is making inroads to large-scale physics projects in general.
There is also the particle accelerator at HZDR that uses open62541.

http://accelconf.web.cern.ch/AccelConf/icalepcs2017/papers/thpha166.pdf

jpfr, 2018-09-21:
I added the reference to the wiki.

closed 2018-09-21 by jpfr