open62541/open62541 #2075: How to read arrays from client

kjeldflarup, 2018-09-20:
I set a server op with this software https://github.com/open62541/open62541/blob/master/examples/tutorial_server_variabletype.c

UAexpert can easily read the values. But how to do it with open62541?

In the tutorials all examples just use this call `UA_Client_readValueAttribute`, but that just returns a NULL object.

Also the documentation is quite weak on this spot.

I do not expect to use it in a product, but will need it to do some performance tests on my server.

jpfr, 2018-09-20:
I dont’t understand where you want to go with this.
Do we miss an API or is the existing functionality broken?

ReadValueAttribute returns a variant.
Variants can hold arrays.

kjeldflarup, 2018-09-21:
While trying to isolate the code that fail, I suddenly nolonger can reproduce the behaviour. So the error must be on my coding.

Anyway thanks for confirming the behaviour.

closed 2018-09-21 by kjeldflarup