open62541/open62541 #2082: sendSymmetricChunk() writes to send buffer without checking its boundaries

jonasgreen88, 2018-09-24:
Description
===========
Last week we added an encrypted endpoint to our server. When we connected to the endpoint we almost immediately encountered a fatal crash of our device. We have now tracked the crash to the stack that writes out of the boundaries of the send buffer in some cases. In our configuration the message chunks are the minimum size 8196 bytes.

The function sendSymmetricChunk() writes both the padding and the signature to the send buffer without verifying that it doesn't exceeds the send buffer. In our case when we encountered the device crash the msg size + padding + signature > 8196 bytes.
https://github.com/open62541/open62541/blob/c9c2088124d6b3ff4e3068921812f1abdac40536/src/ua_securechannel.c#L704

We are planing to test a solution that will verify that the buffer boundaries are not exceeded when writing the padding and the signature and return BADENCODINGLIMITSEXCEEDED if there not enough space. Do you think that's the right solution for this issue?

When returning this error code we think the error will be handled at this point, is that right?
https://github.com/open62541/open62541/blob/c9c2088124d6b3ff4e3068921812f1abdac40536/src/ua_types_encoding_binary.c#L1418
If that is the case, the last added member will be removed and the stack will try to send the chunk again. But imagine that the last member is a small one and just removing that member doesn't free up enough space of the chunk for the padding and the signature. Then the sendSymmetricChunk() function will fail again because of the same reason. In that case the message will be treated as "too large". Shouldn't the stack try to remove some more members from the encoded data before giving up sending the message chunk or maybe even reserve space for the padding and signature before encoding the data to send?

We can help out making a fix for this issue but may need some guiding from you who designed the encoding algorithm to get a good solution.

Background Information / Reproduction Steps
===========================================
Run the stack with message chunk size 8196 bytes and connect to a SignAndEncrypt endpoint.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 43246510c0803dfd24fb42bf9a4e0e4a90679a96 (0.3 branch)
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jonasgreen88, 2018-09-25:
Forget about the solution I posted yesterday. This morning I found the function setBufPos() which reserves space for the signature and the padding. But when it reserved space for the header in the send buffer, it also moved the end of the buffer with the same length as the header. It also did not reserve enough space for the worse-case padding. I will post a PR in a while.
https://github.com/open62541/open62541/blob/c9c2088124d6b3ff4e3068921812f1abdac40536/src/ua_securechannel.c#L572

jpfr, 2018-09-25:
> Forget about the solution I posted yesterday. This morning I found the function setBufPos() which reserves space for the signature and the padding.

That's a hint that maybe we could add better documentation of the code structure for the SecureChannel. Like a small ASCII-graphics of the chunk structure. And which methods deal with which part of the chunk.

jpfr, 2018-09-27:
Closed via #2084.

closed 2018-09-27 by jpfr