open62541/open62541 #2086: Could we have example of composing "full-featured" event/alarm

coder4803, 2018-09-27:
I am trying to compose standard "OffNormalAlarmType" event (http://documentation.unified-automation.com/uasdkcpp/1.5.4/html/Doc_OpcUa_AlarmsAndConditions_ConditionTypes.html#Doc_OpcUa_OffNormalAlarmType).

The event consists of methods and Non-trivial variable fields (ActiveState/TwoStateVariableType, http://documentation.unified-automation.com/uasdkcpp/1.5.4/html/Doc_OpcUa_AlarmsAndConditions_VariableTypes.html#Doc_OpcUa_TwoStateVariableType). I have troubles adding either of these fields (trivial-variables work as described in server_tutorial_events.c).

I am monitoring events with UaExpert. I get this OffNormalAlarmType event and trivial-variable fields (i.e. Severity) are set as expected but other fields have some "default/non-existing" values.

The whole API looks quite professional but I still couldn't find this out.

sqadasi, 2018-09-27:
Alarms are not supported yet by the server, however, you still could monitor an Event from OffNormalAlarmType in EventView of UaExpert.
ActiveState is mandatory Field of AlarmConditionType, therefore that Field should be added automatically to your Event Node during its creation. The requirement for that behaviour, is that the Part 9 NodeSet is integrated in the server.
hope that shed some light on your problem...

ccvca, 2018-09-27:
#2083

coder4803, 2018-10-01:
If I understood right, it is not possible to generate valid OffNormalAlarmType event/alarm with current master-API. But is it still possible to generate custom event that contains TwoStateVariableType field (and it's properties)? I tried to do this with "UA_Server_addVariableNode" as with objects but had no luck.

jpfr, 2018-10-01:
> I tried to do this with "UA_Server_addVariableNode" as with objects but had no luck.

That should work.
Events are represented as objects in the information model before they are triggered.

Does UA_Server_addVariableNode return anything?
Otherwise you might have to add a custom EventType so that the client "sees" the new field that can be selected for the event payload.

Pro, 2018-11-06:
@coder4803 any update on this? Is the issue solved for you?

teccheck, 2025-04-15:
As there has been no more activity and examples available now, I'll close this

closed 2025-04-15 by teccheck