open62541/open62541 #2089: specification for variable callback or datasource

fabian727, 2018-10-02:
Hi guys,

I am sorry for bothering you, but I searched the spec for a whole hour and didn't find any information.

Where is it defined, how to access (read/write) a variable?
you implemented like any other lib the three ways: manually, callback, datasource. [I refer to those](https://open62541.org/doc/current/tutorial_server_datasource.html)

In which part of the specification are those mentioned, if so?

Thanks,

Fabian

jpfr, 2018-10-02:
They are not mentioned at all.
The spec defines "the protocol", i.e. the behaviour that is visible over the network.

So the implementation of the data source is not covered by the spec.
Does that answer the question?

jpfr, 2018-10-02:
You can of course use the *Read* and *Write* service to access the variables (over the network).

fabian727, 2018-10-03:
thanks, yes that does answer my question.

how does it come, that you have the same principles and working with the variables as UnifiedAutomation?
you can manually read/write, you have the options of callback and data source; exact same as UA.

jpfr, 2018-10-03:
Uhm. No idea. Maybe because these are the principles that make sense?
Which of the Unified SDK are you referring to?

My guess is that it's pretty similar for Softing and Matrikon as well.
Callbacks are pretty common for C programming.

fabian727, 2018-10-04:
I am not referring the Callback as the C implementation,
but the three ways of "grabbing the data "
- read/write the variable
- callback (there is a C-Function, but the value will be saved in the variable itself
-datasource (there is a C-Function, but the programmer has to care about the saving of the variable)

I am refering to the UA C++ Library. The handling there is more complex than needed (in my opinion)., but there are those three ways to handle with a variable too, plus they generate the time marks as you do. If I grab a variable by callback/ datasource the generation and server time mark are identical.

jpfr, 2018-10-04:
Ok. Looking at their C++ API, it seems you have to implement a custom "NodeManager" whenever you want to have custom access to a node.
It seems it is not possible to attach callbacks directly to nodes in the Unified C++ API.

I would not read too much into that.
There are of course similarities between all vendors. Because some things just make sense.
But in many important ways, we have chosen different designs.
For example, a variant in open62541 can hold any datatype with any dimensionality. And it all goes through the same code path.
Whereas some vendors have near-identical duplicated code-paths for every type of variant member.
And they only support scalar, vector and matrix.

Please reopen if questions remain.

closed 2018-10-04 by jpfr