open62541/open62541 #2090: Cannot create the myServer sample from the tutorial

effiko, 2018-10-02:
**Compile time error** 
c:\test\open62541>gcc --version
gcc (MinGW.org GCC-6.3.0-1) 6.3.0
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

c:\test\open62541>gcc -std=c99 open62541.c myServer.c -lws2_32 -o myServer
open62541.c: In function 'addServerSocket':
open62541.c:38913:42: error: 'IPV6_V6ONLY' undeclared (first use in this function)
        setsockopt(newsock, IPPROTO_IPV6, IPV6_V6ONLY,
                                          ^~~~~~~~~~~
open62541.c:38913:42: note: each undeclared identifier is reported only once for each function it appears in
open62541.c: In function 'UA_DateTime_localTimeUtcOffset':
open62541.c:39523:5: warning: implicit declaration of function 'gmtime_s' [-Wimplicit-function-declaration]
     gmtime_s(&ptm, &rawtime);
     ^~~~~~~~

jpfr, 2018-10-02:
We use Mingw-w64 for our continuous integration builds.
That is not the same as mingw.org.

https://mingw-w64.org/doku.php/start
https://sourceforge.net/projects/mingw-w64/files/Toolchains%20targetting%20Win32/Personal%20Builds/mingw-builds/installer/mingw-w64-install.exe/download

They have more complete support for the Winsock2 API.
Does that work for you?

jpfr, 2018-10-22:
Please reopen if required.

closed 2018-10-22 by jpfr