open62541/open62541 #2092: Question: Subscription / Monitoring - Connection Drop

track02, 2018-10-03:
I'm currently running a FreeOpcUa server on a Raspberry Pi and connecting to it with a Qt Android Application (QtOpcUa + Open62541 Backend)

I seem to have some trouble monitoring nodes for value changes as the connection will just drop out after around 10 minutes of starting the monitoring.

I'm having a hard time debugging this and was just wondering if this is something you've came across before or would have any better ideas on how to diagnose.

I enabled logging on my server and can see an entry closing the connection to the client and cleaning up. Looking at the output the application provides only shows the following from open62541:

`qt.opcua.plugins.open62541: Unable to send publish request`

I've tried Wireshark but can't seem to spot anything as the packets stop when the connection drops, this may just be my inexperience using it so please point me in the right direction if possible.

I have also tried changing the log level when building open62541 but don't see anything new (again this may be an error on my behalf).

Any thoughts or ideas would be greatly appreciated.

Many thanks.

ChHaa, 2018-10-05:
Did you change the def. values for session lifetime ?

jpfr, 2018-10-22:
Hello @track02 ,

do you have log messages that were generated when the disconnect happened?

track02, 2018-10-22:
Hi,

Unfortunately don't have them to hand but can try and recreate the drop-out tomorrow if you'd like.
From memory the client would log a warning from Open62541 like this:

Open62541: Unable to Send Publish Request

And the server would just log a client disconnection and cleanup of the session.

I've tried using a Node.js server instead and have no issues but I have noticed logs about security tokens changing with a timeout of 600000ms (10 min) so I wonder if there's something similar going on with the Python FreeOpcUa server.

closed 2025-04-15 by teccheck