open62541/open62541 #2093: UA_Client_MonitoredItems_createDataChange won't monitor the NodeID supplied

smokingaces88, 2018-10-04:
Description
===========
I am having problems using the UA_Client_MonitoredItems_createDataChange function. I have included the code background but in summary I create a monitored items request then call the mentioned function but for some reason when I dig into the UA_MonitoredItemCreateResult response it seems to drop the nodeID I pass it and sets to zero (in effect setting nodeID 0 as the monitored item).

Background Information / Reproduction Steps
===========================================
My application code is as follows:

```
//Create the Subscription:
	UA_CreateSubscriptionRequest request = UA_CreateSubscriptionRequest_default();
	UA_CreateSubscriptionResponse response = UA_Client_Subscriptions_create(client, request, NULL, NULL, NULL);
	UA_Int32 subId = response.subscriptionId;

	UA_MonitoredItemCreateRequest monRequest = UA_MonitoredItemCreateRequest_default(UA_NODEID_NUMERIC(0, node));
	monRequest.monitoringMode = UA_MONITORINGMODE_REPORTING;
	monRequest.requestedParameters.samplingInterval = 100;
	monRequest.requestedParameters.queueSize = 1;
	UA_MonitoredItemCreateResult monResponse = 
		UA_Client_MonitoredItems_createDataChange(client, subId, UA_TIMESTAMPSTORETURN_BOTH, monRequest, NULL, test_callback, NULL);

	if (monResponse.statusCode == UA_STATUSCODE_GOOD) {
		UA_Int32 id = monResponse.monitoredItemId;
		UA_NodeId node = monResponse.filterResult.content.encoded.typeId;
	}
```

When debugging I can trace where the nodeID gets dropped to within the createDataChange function and more specifically:

```
UA_MonitoredItemCreateResult UA_EXPORT
UA_Client_MonitoredItems_createDataChange(UA_Client *client, UA_UInt32 subscriptionId,
          UA_TimestampsToReturn timestampsToReturn, const UA_MonitoredItemCreateRequest item,
          void *context, UA_Client_DataChangeNotificationCallback callback,
          UA_Client_DeleteMonitoredItemCallback deleteCallback) {
    UA_CreateMonitoredItemsRequest request;
    `UA_CreateMonitoredItemsRequest_init(&request);
    request.subscriptionId = subscriptionId;
    request.timestampsToReturn = timestampsToReturn;
    request.itemsToCreate = (UA_MonitoredItemCreateRequest*)(uintptr_t)&item;
    request.itemsToCreateSize = 1;
    **_UA_CreateMonitoredItemsResponse response = 
       UA_Client_MonitoredItems_createDataChanges(client, request, &context,
                                                   &callback, &deleteCallback);_**
    UA_MonitoredItemCreateResult result;
    UA_MonitoredItemCreateResult_init(&result);
    if(response.responseHeader.serviceResult != UA_STATUSCODE_GOOD)
        result.statusCode = response.responseHeader.serviceResult;

    if(result.statusCode == UA_STATUSCODE_GOOD &&
       response.resultsSize != 1)
        result.statusCode = UA_STATUSCODE_BADINTERNALERROR;
    
    if(result.statusCode == UA_STATUSCODE_GOOD)
       UA_MonitoredItemCreateResult_copy(&response.results[0] , &result);
    UA_CreateMonitoredItemsResponse_deleteMembers(&response);
    return result;
}
```
I have tried to highlight the line where the issue occurs; it is when the following line is called:    UA_CreateMonitoredItemsResponse response = 
       UA_Client_MonitoredItems_createDataChanges(client, request, &context,
                                                   &callback, &deleteCallback);

The request argument contains the nodeID correctly but the response.results.filterResult.content.encoded.typeId.identifier.numeric = 0. Can anyone suggest what might be causing this problem for me? Or can suggest any further ways I can debug this issue to find further info?

Thanks.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0.3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

smokingaces88, 2018-10-04:
Update: I managed to successfully create the monitored item by doing the following and then calling the createDataChange function:
UA_NodeId nodeid = UA_NODEID_NUMERIC(0, node);

I now have the problem in which the DataChangeNotificationCallback is not getting called on the data value changing. In addition to the above can anybody suggest if any further configuration is required for the callback? 

For reference the function I pass is "test_callback" which is a static function:
`static void test_callback(UA_Client *client, UA_UInt32 subId, void *subContext,
	UA_UInt32 monId, void *monContext, UA_DataValue *value) {
	UA_UInt32 sub = subId;
}`

I placed my breakpoint in for test purpose but it never seems to get called.

jpfr, 2018-10-04:
The client does not open a thread.
Do you call `UA_Client_run_iterate` from time to time?
Otherwise the client cannot perform the background tasks. Like handling subscription responses.

jpfr, 2018-10-04:
This unit test covers your case:
https://github.com/open62541/open62541/blob/master/tests/client/check_client_subscriptions.c#L108

smokingaces88, 2018-10-08:
Thank you, did the trick.

closed 2018-10-08 by smokingaces88