open62541/open62541 #2094: Client - HowTo request multiple items with UA_Client_MonitoredItems_createDataChange

jrcoding, 2018-10-04:
Description
===========
I was wondering if someone has a client example of how to request multiple items with one call to UA_Client_MonitoredItems_createDataChange.

Background Information / Reproduction Steps
===========================================

Digging into the code of ua_client_subscriptions.c I saw that in function UA_Client_MonitoredItems_createDataChange the structurevariable  "request.itemsToCreateSize" is hardcoded initialized to 1 which means, as far as I understand, that only one item could be requested at a time calling UA_Client_MonitoredItems_createDataChange, right?

Because every call to UA_Client_MonitoredItems_createDataChange leads to a PublishRequest/PublishResponse network transfer I would like to request a bunch of items within one call.

Any advice is welcome...

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

StalderT, 2018-10-04:
You may use test 

https://github.com/open62541/open62541/blob/8c5eb993bf3e2a5397ec8bf380fcfe91eafc085b/tests/client/check_client_subscriptions.c#L109-L204

jrcoding, 2018-10-05:
Thanks a lot, that's what I was looking for....

closed 2018-10-05 by jrcoding