open62541/open62541 #2102: Server Crash when updating tag value with 10ms interval

lisantanggit, 2018-10-09:
Description
===========
I test the throughput capability of open62541.
However, in my test case, the Server will crash after about 5 hours.

Background Information / Reproduction Steps
===========================================
Test environment 

- description:

Building environment
Ubuntu16.04,
GCC 5.4
open62541 0.3-rc2

Building 

- steps

Configure using default options, without using UA_ENABLE_MULTITHREADING
cmake && make && make install

- Test  case(Server)

Create 1 tags(INT type) at Server
Create 1 thread to modify the corresponding tags with 10ms time interval 

- Test case (Client)
Run 2 UA Expert at the PC where Server runs
Run 3 UA Export at another PC (connected with 1Gb router)

- Test results

Server will crash when performing function UA_NodeMap_releaseNode()

- Stack trace information of core file
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `./open62541_test'.
Program terminated with signal SIGABRT, Aborted.
#0  0x00007f6ae6cbd428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54      ../sysdeps/unix/sysv/linux/raise.c: 没有那个文件或目录.
[Current thread is 1 (Thread 0x7f6ae697e700 (LWP 25941))]
(gdb) bt
#0  0x00007f6ae6cbd428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007f6ae6cbf02a in __GI_abort () at abort.c:89
#2  0x00007f6ae6cb5bd7 in __assert_fail_base (fmt=<optimized out>, assertion=assertion@entry=0x473475 "entry->refCount > 0", 
    file=file@entry=0x4733f8 "/media/siemens/ws/02_OPCUA/01_OSS_OPC_UA/04_open62541/01_env/open62541/plugins/ua_nodestore_default.c", 
    line=line@entry=280, function=function@entry=0x4734f0 <__PRETTY_FUNCTION__.7774> "UA_NodeMap_releaseNode") at assert.c:92
#3  0x00007f6ae6cb5c82 in __GI___assert_fail (assertion=0x473475 "entry->refCount > 0", 
    file=0x4733f8 "/media/siemens/ws/02_OPCUA/01_OSS_OPC_UA/04_open62541/01_env/open62541/plugins/ua_nodestore_default.c", line=280, 
    function=0x4734f0 <__PRETTY_FUNCTION__.7774> "UA_NodeMap_releaseNode") at assert.c:101
#4  0x000000000045c161 in UA_NodeMap_releaseNode (context=0x1e2c270, node=0x1e52c20)
    at /media/siemens/ws/02_OPCUA/01_OSS_OPC_UA/04_open62541/01_env/open62541/plugins/ua_nodestore_default.c:280
#5  0x000000000042ca03 in UA_Server_editNode (server=0x1e2dc70, session=0x689220 <adminSession>, nodeId=0x7f6ae697dd20, 
    callback=0x4321dc <copyAttributeIntoNode>, data=0x7f6ae697dd20)
    at /media/siemens/ws/02_OPCUA/01_OSS_OPC_UA/04_open62541/01_env/open62541/src/server/ua_server_utils.c:363
#6  0x0000000000433182 in UA_Server_write (server=0x1e2dc70, value=0x7f6ae697dd20)
    at /media/siemens/ws/02_OPCUA/01_OSS_OPC_UA/04_open62541/01_env/open62541/src/server/ua_services_attribute.c:1306
#7  0x000000000043329e in __UA_Server_write (server=0x1e2dc70, nodeId=0x7f6ae697ddf0, attributeId=UA_ATTRIBUTEID_VALUE, 
    attr_type=0x6828b0 <UA_TYPES+1104>, attr=0x7f6ae697de08)
    at /media/siemens/ws/02_OPCUA/01_OSS_OPC_UA/04_open62541/01_env/open62541/src/server/ua_services_attribute.c:1332
#8  0x00000000004019cc in UA_Server_writeValue (server=0x1e2dc70, nodeId=..., value=...)
    at /media/siemens/ws/02_OPCUA/01_OSS_OPC_UA/04_open62541/01_env/open62541/build/open62541.h:11432
#9  0x0000000000401f71 in thread_entry (arg=0x1e4ee50) at ../server.cpp:90
#10 0x00007f6ae70596ba in start_thread (arg=0x7f6ae697e700) at pthread_create.c:333
#11 0x00007f6ae6d8f41d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
(gdb) 
```

-Server Code
```
# include "open62541.h"
#include <signal.h>

/* Build Instructions (Linux)
 * - gcc -std=c99 -c open62541.c
 * - g++ server.cpp open62541.o -o server */

using namespace std;

#define THREAD_NUMBNER		1

UA_Boolean running = true;
UA_Logger logger = UA_Log_Stdout;

static void stopHandler(int sign) {
    UA_LOG_INFO(logger, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}
#include <pthread.h>
#include <stdlib.h>
#include <unistd.h>
#include <string.h>
#include <stdio.h>

typedef struct thread_args{
	UA_Server 	*server_;
	int			thread_index_;
}Thread_Args_T;

void create_vars(UA_Server *server, int index)
{
    // add a variable node to the adresspace
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    UA_Int32 myInteger = index * 100;
    UA_Variant_setScalarCopy(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);

    char buf[64];
    memset(buf, 0, 64);
    sprintf(buf, "%s %d", "the answer", index);
    attr.description = UA_LOCALIZEDTEXT_ALLOC("en-US",buf);
    attr.displayName = UA_LOCALIZEDTEXT_ALLOC("en-US",buf);

    memset(buf, 0, 64);
    sprintf(buf, "%s.%d", "the answer", index);
    UA_NodeId myIntegerNodeId = UA_NODEID_STRING_ALLOC(1, buf);
    UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME_ALLOC(index, "the answer");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId,
                              parentReferenceNodeId, myIntegerName,
                              UA_NODEID_NULL, attr, NULL, NULL);

    /* allocations on the heap need to be freed */
    UA_VariableAttributes_deleteMembers(&attr);
    UA_NodeId_deleteMembers(&myIntegerNodeId);
    UA_QualifiedName_deleteMembers(&myIntegerName);
}

void *thread_entry(void *arg)
{
	Thread_Args_T *parsed = (Thread_Args_T *)arg;
	UA_Server *server = parsed->server_;
	int index = parsed->thread_index_;

	char buf[64];
	memset(buf, 0, 64);
    sprintf(buf, "%s.%d", "the answer", index);
    UA_NodeId myIntegerNodeId = UA_NODEID_STRING(1, buf);

    int sleep_delta = (rand() % 90) + 30;
    int i = 0;
    while (1){

        /* Write a different integer value */
        UA_Int32 myInteger = index * 100 + i; //rand() % 100 | (index <<24);
        UA_Variant myVar;
        UA_Variant_init(&myVar);
        UA_Variant_setScalar(&myVar, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
        UA_Server_writeValue(server, myIntegerNodeId, myVar);

        i++;
        if (i >= 100){
        	i = 0;
        }

    	usleep(1000 * 10);
    }
}

int main() {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_Server *server = UA_Server_new(config);

    for (int i = 0; i < THREAD_NUMBNER; i++){
    	create_vars(server, i);
    }

    for (int i = 0; i < THREAD_NUMBNER; i++){
    	pthread_t 	pid;
    	Thread_Args_T *args = (Thread_Args_T *)malloc(sizeof(Thread_Args_T));
    	args->server_ = server;
    	args->thread_index_ = i;
    	pthread_create(&pid,
    			NULL,
				thread_entry,
				(void *)args);

    	pthread_detach(pid);
    }

    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return retval;
}

```
Checklist
=========
[corefile_stacktrace.txt](https://github.com/open62541/open62541/files/2458932/corefile_stacktrace.txt)

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-10-09:
The library is not thread-safe.
It looks like you access the server from two parallel threads.
In that case, you have to protect the server with a mutex.

Multithreading support can be activated with a compile flag.
But the feature is marked as experimental.

You can add the value update as a "repeated callback" to the server.
So you don't need a second thread.

lisantanggit, 2018-10-10:
Thanks a lot for your information

jpfr, 2018-10-15:
Adding a tag `multithreading` and closing as duplicate.
So we can give notice when multithreading support has been improved.

closed 2018-10-15 by jpfr