open62541/open62541 #2105: unistd.h not found while compiling with clang under Windows

escherstair, 2018-10-09:
Description
===========
I've just tried to use open62541 0.3-rc2 with a compiler which is a porting of clang under Windows 7.
I can' compile my code because I get the error <unistd.h> not found.
This come from amalgamation of
"/home/travis/build/open62541/open62541/src/server/ua_services_discovery.c"
where I can see
```
#ifdef _MSC_VER
# ifndef UNDER_CE
#  include <io.h> //access
#  define access _access
# endif
#else
# include <unistd.h> //access
#endif

```
I've been compiling under Windows, but not with MSC compiler and so <unistd.h> is requested to have `access()` function.

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0.3-rc2
 - [ ] Other OPC UA SDKs used (client or server): none
 - [ ] Operating system: Windows 7 x64
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-10-15:
If you don't use the semaphore file feature, you can wrap the includes with an `#ifdef UA_ENABLE_DISCOVERY_SEMAPHORE`.

If that works for you, feel free to send us a patch for the 0.3 branch.

The includes have already changed on master.

escherstair, 2018-10-17:
Probably it's better to change the first `#define` from `#ifdef _MSC_VER` to `#ifdef _WIN32`.

Pro, 2018-10-22:
Related to #2116

commit 889b7d7f (2018-11-02): Do not include unistd.h on Win32. Fixes #2105