open62541/open62541 #2106: Subscription and Monitor with sampling times under 5ms

schotte, 2018-10-09:
Description
===========
A Client Server connection (both with open62541) is setup and 1 Subscription is established and multiple Monitors (around 5 atm) for this 1 Subscription. The Configuration of the Subscriptions requestPublishingInterval and the monitors samplingInterval is configured. If they are set to 5 ms, it works fine. If it is set to anything below, I am getting a error message from the Client
> error/client	Inactivity for Subscription 1.

I don't really know why this is happening, but it seems like the communication is messed up, they don't communicate for some seconds (see the wireshark attachment (Recorded with 4ms configured) 

If I should supply any other information, please ask. If you have a idea which configuration parameters might limit this, please tell me.
Best Regards Schotte 

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.3-rc2-345-g9921548
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 18.04
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: Loglevel 100
[subscription.log](https://github.com/open62541/open62541/files/2460682/subscription.log)

 - [x] Wireshark network dump attached 
[subscription_inactivity.pcapng.gz](https://github.com/open62541/open62541/files/2460658/subscription_inactivity.pcapng.gz)
 - [ ] Self-contained code example attached
 - [ ] Critical issue

StalderT, 2018-10-13:
@schotte can you provide a sample server and client code?

StalderT, 2018-10-14:
I can reproduce the problem. I'm on it.

StalderT, 2018-10-14:
A timer can  not have a interval < 5 ms : 

https://github.com/open62541/open62541/blob/cce2d622390be4dafd10ebde80fc2dbb3abadc78/src/ua_timer.c#L98-L100

Then the subscription and the monitored item can't be created. The server does not check the result of UA_Timer_addRepeatedCallback.  https://github.com/open62541/open62541/pull/2111 fix it.

jpfr, 2018-10-15:
Closed via #2111.

PR #2111 (2018-10-15): check the result of MonitoredItem_registerSampleCallback and setSubsc…
…riptionSettings

fix #2106
PR #2112 (2018-10-17): add comments about the duration
also fix #2106
closed 2018-10-15 by jpfr