open62541/open62541 #2114: generating multiple events successively fails

sqadasi, 2018-10-15:
Description
===========
Hello everyone!
I've tried to generate multiple events successively in a loop using the events tutorial:
```
static UA_StatusCode
generateEventMethodCallback(UA_Server *server,
                         const UA_NodeId *sessionId, void *sessionHandle,
                         const UA_NodeId *methodId, void *methodContext,
                         const UA_NodeId *objectId, void *objectContext,
                         size_t inputSize, const UA_Variant *input,
                         size_t outputSize, UA_Variant *output) {

    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Creating event");

    /* set up event */
	for(size_t i=0; (i<5)&&(retval==UA_STATUSCODE_GOOD); i++) {
	  UA_NodeId eventNodeId;
	  UA_StatusCode retval = setUpEvent(server, &eventNodeId);
	  if(retval != UA_STATUSCODE_GOOD) {
	  UA_LOG_WARNING(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
		  			  "Creating event failed. StatusCode %s", UA_StatusCode_name(retval));
	  return retval;
	  }

	retval = UA_Server_triggerEvent(server, eventNodeId, UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER), NULL, UA_TRUE);
	if(retval != UA_STATUSCODE_GOOD)
		UA_LOG_WARNING(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, 
						  "Triggering event failed. StatusCode %s", UA_StatusCode_name(retval));
	}
    return retval;
}
```

however, I've faced two problems:
1) only the first event doesn't show normal field values. Other Events are ok.
[image]

2) when generating the events for the second time, or when trying to set event filter with UaExpert, then the server crashes. The assert check fails at: https://github.com/open62541/open62541/blob/271c944049d0b33435f43ced139f5d76fcafadc3/src/server/ua_subscription_monitoreditem.c#L273

I couldn't find the root of the problem, so your help is appreciated :)

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 271c944049d0b33435f43ced139f5d76fcafadc3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

aDogCalledSpot, 2018-10-16:
Unfortunately, I'm unable to reproduce these issues locally, could you maybe provide a Wireshark log where you connect, create the monitored item, call the method and then either modify the filter or call it again (perhaps both would be best) and then I can try finding out what's going on.

sqadasi, 2018-10-16:
@aDogCalledSpot thank you for responding!
I hope those are logs which you need:

1) Open SecureChannelResponse
```
Frame 148: 189 bytes on wire (1512 bits), 189 bytes captured (1512 bits) on interface 0
    Interface id: 0 (\Device\NPF_{5295E406-3F56-45E1-9A83-302580141ACC})
        Interface name: \Device\NPF_{5295E406-3F56-45E1-9A83-302580141ACC}
    Encapsulation type: Ethernet (1)
    Arrival Time: Oct 16, 2018 11:35:08.615306000 Mitteleuropäische Sommerzeit
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1539682508.615306000 seconds
    [Time delta from previous captured frame: 0.019707000 seconds]
    [Time delta from previous displayed frame: 0.019707000 seconds]
    [Time since reference or first frame: 40.142613000 seconds]
    Frame Number: 148
    Frame Length: 189 bytes (1512 bits)
    Capture Length: 189 bytes (1512 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: eth:ethertype:ip:tcp:opcua]
    [Coloring Rule Name: TCP]
    [Coloring Rule String: tcp]
Ethernet II, Src: Hilscher_2c:b7:3d (00:02:a2:2c:b7:3d), Dst: Luxshare_0e:ee:05 (3c:18:a0:0e:ee:05)
    Destination: Luxshare_0e:ee:05 (3c:18:a0:0e:ee:05)
        Address: Luxshare_0e:ee:05 (3c:18:a0:0e:ee:05)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Source: Hilscher_2c:b7:3d (00:02:a2:2c:b7:3d)
        Address: Hilscher_2c:b7:3d (00:02:a2:2c:b7:3d)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Type: IPv4 (0x0800)
Internet Protocol Version 4, Src: 192.168.0.20, Dst: 192.168.0.10
    0100 .... = Version: 4
    .... 0101 = Header Length: 20 bytes (5)
    Differentiated Services Field: 0x00 (DSCP: CS0, ECN: Not-ECT)
        0000 00.. = Differentiated Services Codepoint: Default (0)
        .... ..00 = Explicit Congestion Notification: Not ECN-Capable Transport (0)
    Total Length: 175
    Identification: 0x000a (10)
    Flags: 0x0000
        0... .... .... .... = Reserved bit: Not set
        .0.. .... .... .... = Don't fragment: Not set
        ..0. .... .... .... = More fragments: Not set
        ...0 0000 0000 0000 = Fragment offset: 0
    Time to live: 64
    Protocol: TCP (6)
    Header checksum: 0xf8d0 [validation disabled]
    [Header checksum status: Unverified]
    Source: 192.168.0.20
    Destination: 192.168.0.10
Transmission Control Protocol, Src Port: 4840, Dst Port: 52077, Seq: 29, Ack: 194, Len: 135
    Source Port: 4840
    Destination Port: 52077
    [Stream index: 1]
    [TCP Segment Len: 135]
    Sequence number: 29    (relative sequence number)
    [Next sequence number: 164    (relative sequence number)]
    Acknowledgment number: 194    (relative ack number)
    0101 .... = Header Length: 20 bytes (5)
    Flags: 0x018 (PSH, ACK)
        000. .... .... = Reserved: Not set
        ...0 .... .... = Nonce: Not set
        .... 0... .... = Congestion Window Reduced (CWR): Not set
        .... .0.. .... = ECN-Echo: Not set
        .... ..0. .... = Urgent: Not set
        .... ...1 .... = Acknowledgment: Set
        .... .... 1... = Push: Set
        .... .... .0.. = Reset: Not set
[... 1379 lines cut ...]
                                            Variant Type: Null (0x00)
                                        [11]: Variant
                                            Variant Type: Null (0x00)
                                        [12]: Variant
                                            Variant Type: Null (0x00)
                                        [13]: Variant
                                            Variant Type: Null (0x00)
                                        [14]: Variant
                                            Variant Type: Null (0x00)
                                        [15]: Variant
                                            Variant Type: Null (0x00)
                                        [16]: Variant
                                            Variant Type: Null (0x00)
                                        [17]: Variant
                                            Variant Type: Null (0x00)
                                        [18]: Variant
                                            Variant Type: Null (0x00)
            Results: Array of StatusCode
                ArraySize: -1
            DiagnosticInfos: Array of DiagnosticInfo
                ArraySize: -1

```

9) When calling the function for the second time, there is no response because the server crashes. The same when changing the filter, there is no response after ModifyMonitoredItemsRequest.

sqadasi, 2018-10-19:
@aDogCalledSpot ,I tested further and noticed the following: 
1) when changing the max MonitoredItem queue size from 1 to a number greater than the number of fired events, then it works without triggering the overflow event. (**so it could be the fix for this issue**)

2) However, when leaving max MonitoredItem queue size = 1, and triggering only one event at time with one method call (like in the toturial, but event nodes won't be deleted after triggering the event) there are two results:
a) when calling the method successively in a slow manner, no overflow event is generated.
b) when calling the method successively in a fast manner, the overflow event is generated and the server crashes (fails in the mentioned assert).
I thought it maybe dependent on the sampling interval, but setting it to a low value (5ms) in server and client won't help.

What could be the cause of this behaviour?

aDogCalledSpot, 2018-10-26:
I'm still having trouble reproducing the issue. Have you tried this with an own client? 

Could you send me your full code for the server, just to be sure nothing's missing.

sqadasi, 2018-10-30:
@aDogCalledSpot thank you for responding, Ari!
I am using an embedded device as server with own operating system, therefore think it won't help sending you the code...
However, I have tested further, this time with default server configuration and could reproduce the error (so I think you too could reproduce the error this way :))
1) change the monitoredItem queue size limits range to (1, 1) or other small value so you can generate the overflow event in
https://github.com/open62541/open62541/blob/e229a3514a79edba15ecd5abcb9dbca99fcfaf49/plugins/ua_config_default.c#L275

2) use the toturial code provided in my first comment to generate multiple events
https://github.com/open62541/open62541/issues/2114#issue-370234772

3) first time calling the generating method, an overflow event will be generated plus a number of (queueSize - 1) normal events

4) second time calling the generating method will crash the server

I checked the function 
https://github.com/open62541/open62541/blob/e229a3514a79edba15ecd5abcb9dbca99fcfaf49/src/server/ua_subscription_monitoreditem.c#L234
and found that the check in 
https://github.com/open62541/open62541/blob/e229a3514a79edba15ecd5abcb9dbca99fcfaf49/src/server/ua_subscription_monitoreditem.c#L235
is rather strange when we have an overflow event, because we could get very large values resulting from "mon->queueSize - mon->eventOverflows", for example when we have 2 Overflows from the first call and the queueSize is 1 from the second call.

As an Idea, I thought I could reset the the overflow event value of the monitoredItem (mon->eventOverflows = 0; ) when dequeuing the events (at the end of the function):
https://github.com/open62541/open62541/blob/e229a3514a79edba15ecd5abcb9dbca99fcfaf49/src/server/ua_subscription_monitoreditem.c#L105

With that change, when the queue size is small, then an overflow event will be generated plus a number of (queueSize) normal events. However the server won't crash on multiple calls. 

I would like to ask, whether this behaviour is correct, or should we never get more events (including the overflow event) than the queue size?

RFil, 2023-02-21:
Try compiling with UA_MULTITHREADING=100

teccheck, 2025-04-15:
This functionality has been tested over multiple years via CI and CTT, so it should work now. As there's been no recent activity, I'll close this

closed 2025-04-15 by teccheck