open62541/open62541 #2115: UA_BrowseRequest_deleteMembers free error

jrcoding, 2018-10-16:
Description
===========
Taking client.c from the examples to browse nodes of the example  server and inititalizing the nodeId field in a UA_BrowseRequest structure with UA_NODEID_STRING results in an free error when calling 
UA_BrowseRequest_deleteMembers. 

Background Information / Reproduction Steps
===========================================

I've attached a small example with comments which reproduces the error

[fail_browse.c.gz](https://github.com/open62541/open62541/files/2482094/fail_browse.c.gz)

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

Pro, 2018-10-17:
`bReq.nodesToBrowse[0].nodeId = UA_NODEID_STRING(1, "the.answer");` will add a reference to the passed string, it will not create a copy.
Since `UA_BrowseRequest_deleteMembers(&bReq);` is trying to delete all of its content, including the nodeId, it will try to free the string `the.answer` but it is on the stack, not heap.

Try to use `bReq.nodesToBrowse[0].nodeId = UA_NODEID_STRING_ALLOC(1, "the.answer");`. This will use malloc to create a copy of the string.

Please reopen, if the issue still remains.

closed 2018-10-17 by Pro

jrcoding, 2018-10-18:
works fine, thank's