open62541/open62541 #2117: UA_BYTESTRING_ALLOC fails on data containing \0

gusentom, 2018-10-17:
Description
===========
UA_BYTESTRING_ALLOC fails if byte data contains a \0 because strlen() terminates on this position and the length of the resulting ByteString is less than the original length.
In the original example I tried to get the data of a png into a ByteString, unfortunately there was a \0 on the eight position.

Background Information / Reproduction Steps
===========================================

  char *val = "asdf\0asdf";

  UA_ByteString byteStr = UA_BYTESTRING_ALLOC(val);
  printf("byteStr.length = %d\n", (int)byteStr.length);
  printf("byteStr.data = %s\n", byteStr.data);

-->
byteStr.length = 4
byteStr.data = asdf�

Checklist
=========
Please provide the following information:

 - [ x ] open62541 Version (release number or git tag): v1.04
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-10-17:
Hello @gusentom ,

that is to be expected. The only we to find out the length of a C-string (char array) is to look for the first \0. Because we can only call `strlen` internally.
If you know the length, you can wrap the pointer in a UA_ByteString and call UA_ByteString_copy.

For me this is WONTFIX.
Please reopen if required.

closed 2018-10-17 by jpfr

gusentom, 2018-10-17:
Hallo Julius,

wir sind dabei den open62541 Server für eines unserer Entwicklungsprojekte
zu evaluieren und haben versucht die unterschiedlichsten Datentypen zu
verwenden.
Beim Versuch eine Variable mit dem Datentyp UA_TYPES_IMAGEPNG in das Modell
einzuhängen und die read Methode zu implementieren bin ich auf dieses
Problem gestoßen.

hier die aktuelle Implementierung mit UA_ByteString_allocBuffer():

static UA_StatusCode readMrLinearValue(UA_Server *server, const UA_NodeId
*sessionId, void *sessionContext,
    const UA_NodeId *nodeId, void *nodeContext, UA_Boolean sourceTimeStamp,
const UA_NumericRange *range,
    UA_DataValue *dataValue)
{
  char* base64DecodeOutput = NULL;
  size_t size;
  Base64Decode(test_data.mrLinear, (unsigned char**) &base64DecodeOutput,
&size);

  char value[size];
  memcpy(&value, base64DecodeOutput, size);
  free(base64DecodeOutput);

  UA_ByteString str;
  UA_ByteString_allocBuffer(&str, size);
  str.length = size; //strlen(value);
  str.data = (UA_Byte*) value;

  UA_Variant_setScalarCopy(&dataValue->value, &str,
&UA_TYPES[UA_TYPES_IMAGEPNG]);
  dataValue->hasValue = true;

  return UA_STATUSCODE_GOOD;
}

diese read Methode führt mich noch zu einem weiteren Problem und zwar kann
der UA Expert den Wert dieser Variable in der Attributansicht lesen, beim
Anmelden dieser Variable
in der Data Access View kommt allerdings der Statuscode
BadWaitingForInitialData -> siehe bitte beiliegenden Screenshot.

ein vergleichbares Beispiel mit dem Testserver von UnifiedAutomation
funktioniert soweit, habe ich in der read Methode etwas falsch
implementiert?

Thomas Gusenleitner, DI (FH)
Software Entwicklung
Software Engineering

ENGEL AUSTRIA GmbH | Ludwig-Engel-Straße 1 | 4311 Schwertberg | Austria
Tel. +43 50 620 4289
thomas.gusenleitner@engel.at | www.engelglobal.com