open62541/open62541 #2118: Client: How to monitor a Structure variable and access the binary encoded ByteString

jrcoding, 2018-10-18:
Description
===========
I was wondering if one has an example client that monitors a Structure variable and howto access the parts of the monitored structure binary encoded as ByteString?

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-10-18:
You can monitor any variable with the same mechanism.
Structure or not.
If your client knows the structure, it will be decoded automatically.
Otherwise you get an ExtensionObject that wraps the binary encoding.
Then you have to manually decode.

jrcoding, 2018-10-25:
First of all thanks for the advice!

it was also my  assumption that I will get back a Variant that encapsulates the ExtensionObject, but the only data I get back is a Variant where the data only holds the "Identifiyer String" of the ExtensionObject. 

I can see, that the packet on the wire will be received o.k. during "connection_recv" but in my case it seems to me the decoding of the packet throws away some information??? 

My assumption was that I get back the whole ExtensionObject including the ByteString. Am I wrong?

I've attached the wireshark dump of the packet 

```
OpcUa Binary Protocol
    Message Type: MSG
    Chunk Type: F
    Message Size: 114
    SecureChannelId: 684046697
    Security Token Id: 1
    Security Sequence Number: 55
    Security RequestId: 5
    OpcUa Service : Encodeable Object
        TypeId : ExpandedNodeId
            NodeId EncodingMask: Four byte encoded Numeric (0x01)
            NodeId Namespace Index: 0
            NodeId Identifier Numeric: ReadResponse (634)
        ReadResponse
            ResponseHeader: ResponseHeader
                Timestamp: Oct 25, 2018 09:59:31.246622900 CEST
                RequestHandle: 5
                ServiceResult: 0x00000000 [Good]
                ServiceDiagnostics: DiagnosticInfo
                    EncodingMask: 0x00
                        .... ...0 = has symbolic id: False
                        .... ..0. = has namespace: False
                        .... .0.. = has localizedtext: False
                        .... 0... = has locale: False
                        ...0 .... = has additional info: False
                        ..0. .... = has inner statuscode: False
                        .0.. .... = has inner diagnostic info: False
                StringTable: Array of String
                    ArraySize: 0
                AdditionalHeader: ExtensionObject
                    TypeId: ExpandedNodeId
                        EncodingMask: 0x00, EncodingMask: Two byte encoded Numeric
                            .... 0000 = EncodingMask: Two byte encoded Numeric (0x0)
                            .0.. .... = has server index: False
                            0... .... = has namespace uri: False
                        Identifier Numeric: 0
                    EncodingMask: 0x00
                        .... ...0 = has binary body: False
                        .... ..0. = has xml body: False
            Results: Array of DataValue
                ArraySize: 1
                [0]: DataValue
                    EncodingMask: 0x05, has value, has source timestamp
                        .... ...1 = has value: True
                        .... ..0. = has statuscode: False
                        .... .1.. = has source timestamp: True
                        .... 0... = has server timestamp: False
                        ...0 .... = has source picoseconds: False
                        ..0. .... = has server picoseconds: False
                    Value: Variant
                        Variant Type: ExtensionObject (0x16)
                        Value: ExtensionObject
                            TypeId: ExpandedNodeId
                                EncodingMask: 0x03, EncodingMask: String
                                    .... 0011 = EncodingMask: String (0x3)
                                    .0.. .... = has server index: False
                                    0... .... = has namespace uri: False
                                Namespace Index: 3
                                Identifier String: TE_"Struct"
                            EncodingMask: 0x01, has binary body
                                .... ...1 = has binary body: True
                                .... ..0. = has xml body: False
                            ByteString: 01000000000000000099050000a01a1f40f002be41
                    SourceTimestamp: Oct 25, 2018 09:59:31.246613600 CEST
            DiagnosticInfos: Array of DiagnosticInfo
                ArraySize: 0
```

jrcoding, 2018-10-26:
sorry, seems I got lost in the OPCUA data structures....

Every thing is fine, found the ExtensionObject wrapped in the Variant...

closed 2018-10-26 by jrcoding