open62541/open62541 #2120: Reading EngineeringUnits using the client library

falknerdominik, 2018-10-18:
Description
===========

I searched the documentation no how to read the EngineeringUnits from a node and did not find anything (Same with EURange). Is it possible to read it or planned in the future?

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.3-rc2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: client on linux x64/ armv7
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Nik625, 2018-10-19:
The required datatypes need to be enumerated in a specific _.txt_ file to be referenced from the included nodeset.xml (see CMakeLists.txt and the similar datatypes_pubsub.txt file). These and more datatypes are supported by Part8 DataAccess, which is currently under development @Hilscher company.

falknerdominik, 2018-10-20:
@Nik625,
thank you for the quick response. What i gather from your answer is that this is currently under development and not really ready for production? @Hilscher is there any information when this is released or a tutorial for the current implementation?

Nik625, 2018-10-22:
A first draft with a pull request will be coming in the next weeks.

jpfr, 2018-10-22:
@falknerdominik ,

You can already access EngineeringUnits with the default client.
But the will remain in binary encoding wrapped in an ExtensionObject.

In order to automatically unpack EngineeringUnits, you need to
- add the datatype to this file: https://github.com/open62541/open62541/blob/master/tools/schema/datatypes_minimal.txt
- regenerate the datatype sources with cmake

No need to wait for the Alarms&Conditions PR.
Even though the PR will make the handling a lot easier.

GeorgeSellner, 2019-03-21:
Does this apply to server as well?
Will editing the 
        open62541/tools/schema/datatypes_minimal.txt
      load on server as well?

#2594

teccheck, 2025-04-15:
It looks to me like a solution is there, and there's no more activity, so I'll close this issue

closed 2025-04-15 by teccheck