open62541/open62541 #2122: Is there a reason why the ValueRank are not described in an enumartion?

mlang-de, 2018-10-19:
At the moment the ValueRank are described with "magic numbers" like -1 for scalar, 1 for one dimension. Is there a reason not to change it to a human easier readable enumeration?

br

jpfr, 2018-10-19:
The list is “open ended” for positive integers. So that is difficult to capture with an enum.

We can of course use macro defines for the most common valueranks.

Do you have a proposal for a naming scheme?

jpfr, 2018-10-19:
Is this more readable?
```c
#define UA_VALUERANK_SCALAR_OR_ONE_DIMENSION    -3
#define UA_VALUERANK_ANY                     -2
#define UA_VALUERANK_SCALAR                  -1
#define UA_VALUERANK_ONE_OR_MORE_DIMENSIONS      0
#define UA_VALUERANK_ONE_DIMENSION             1
#define UA_VALUERANK_TWO_DIMENSIONS            2
#define UA_VALUERANK_THREE_DIMENSIONS          3
```

Yeah. We need a PR for everything now.
Because we check CLAs with PRs as well.

This could be merged right away without discussion.

mlang-de, 2018-10-19:
I am fine with both variants.
Anyway, I will take yours.

closed 2018-10-19 by mlang-de