open62541/open62541 #2134: ua_network_pubsub_ethernet.c triggers gcc bug 53119

crosland, 2018-10-23:
Description
===========
Line 157 of ua_network_pubsub_ethernet.c
    struct ifreq ifreq = { 0 };
Triggers this gcc bug https://gcc.gnu.org/bugzilla/show_bug.cgi?id=53119 or maybe https://gcc.gnu.org/bugzilla/show_bug.cgi?id=80454 

[ 65%] Building C object CMakeFiles/open62541-object.dir/open62541.c.o
/home/acroslan/smart-factory/open62541/0.4/open62541/build/open62541.c: In function \u2018UA_PubSubChannelEthernet_open\u2019:
/home/acroslan/smart-factory/open62541/0.4/open62541/build/open62541.c:56356:12: error: missing braces around initializer [-Werror=missing-braces]
     struct ifreq ifreq = { 0 };
            ^
/home/acroslan/smart-factory/open62541/0.4/open62541/build/open62541.c:56356:12: error: (near initialization for \u2018ifreq.ifr_ifrn\u2019) [-Werror=missing-braces]
/home/acroslan/smart-factory/open62541/0.4/open62541/build/open62541.c:56356:12: error: missing initializer for field \u2018ifr_ifru\u2019 of \u2018struct ifreq\u2019 [-Werror=missing-field-initializers]
In file included from /home/acroslan/smart-factory/open62541/0.4/open62541/build/open62541.h:147:0,
                 from /home/acroslan/smart-factory/open62541/0.4/open62541/build/open62541.c:28:
/usr/include/net/if.h:149:9: note: \u2018ifr_ifru\u2019 declared here
       } ifr_ifru;
         ^

Background Information / Reproduction Steps
===========================================
Building amalgamation .c/.h after enabling pub/sub in ccmake

Workaround
========
    struct ifreq ifreq = { {{0}}, {{0}} };

Checklist
=========
Please provide the following information:

 - [master branch Git-Revision: 0.3-rc2-443-ga42a5f5 ] open62541 Version (release number or git tag):
 - [n/a ] Other OPC UA SDKs used (client or server): 
 - [ CentOS 7.5.1804] Operating system:
 - [ n/a] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ n/a] Wireshark network dump attached
 - [ n/a] Self-contained code example attached
 - [ No] Critical issue

hthiery, 2018-10-23:
shouldn't we do a memset() instead?

jpfr, 2018-10-23:
> shouldn't we do a memset() instead?

Yes. Especially since the structure is OS-specific.
So there might be differences between architectures.

crosland, 2018-10-23:
struct ifreq ifreq;
    memset(&ifreq, 0, sizeof(struct ifreq));

Also builds just fine :)

Pro, 2018-11-05:
Fixed in #2172

closed 2018-11-05 by Pro